s60sc/ESP32-CAM_MJPEG2SD #137: Access point

fr508, 2022-10-10:
Can your ESP32-CAM_MJPEG2SD app work without an Internet connection like make it become an access point( AP) instead of a station? I mean a connection like this project: https://randomnerdtutorials.com/esp32-cam-access-point-ap-web-server/

s60sc, 2022-10-10:
It defaults to AP

fr508, 2022-10-11:
I mean stream without connecting to a wifi router but I don't need to do that anymore, however thanks for your response!
I get a new issue! After testing through the night, I release that it only saves to the SD card when it detects moving objects. However, I want it to save anything to the SD card like a GoPro or an IP camera

s60sc, 2022-10-11:
AP doesnt use wifi router.
To permanently record, declare a pir pin and keep it high

closed 2022-10-20 by s60sc