s60sc/ESP32-CAM_MJPEG2SD #139: Using a wide angle lens

YapCK45, 2022-10-24:
I recently bought a wide angle lens (160 deg) and replaced the standard lens (70 deg) . Cannot get the motion detection to work. Seems like when everything becomes smaller, the parameters for motion detection need to be tweaked. Would appreciate it if there is some explanation of these parameters.

s60sc, 2022-10-25:
use _Show Motion_ feature to monitor, increase _Motion sensitivity_, reduce _Pixel difference to indicate change_ under _Motion_ in _Edit Config_ tag

YapCK45, 2022-10-26:
I have tried that. _Pixel difference_ is quite intuitive. I have reduced _Num changed frames to start motion_ and increased _Checks per second for start motion_.  I reckoned that cannot be more than the fps setting in Camera Control.
What are _Bottom band where motion is checked_ and _Top band where motion is checked_?  Are they related to _Total number of detection bands_?
This is my configuration:
[image]

closed 2023-01-30 by s60sc