s60sc/ESP32-CAM_MJPEG2SD #141: Password

enord, 2022-10-31:
This was working on one network and I attempted to change to a different wifi network.
I suffered some sort of corruption whilst saving.
The address of 192.168.4.1 cannot be accessed as the wifi connection requires a password but I do not have it.
I have tried re-uploading the file to the ESP32-CAM but it still does not allow me to connect to its wifi server.

[00:00:00.878 updateAppStatus] Enabling motion detection
[00:00:00.940 startWifi] Setting wifi hostname: ESP-CAM_MJPEG_D27EB994
[00:00:00.940 startWifi] No stored Credentials. Starting Access point
[00:00:01.674 setWifiAP] Created Access Point with SSID: ESP-CAM_MJPEG_D27EB994
[00:00:01.675 setWifiAP] Use 'http://192.168.4.1' to connect
[00:00:01.707 setupMndsHost] mDNS service: http://ESP-CAM_MJPEG_D27EB994.local

s60sc, 2022-10-31:
192.168.4.1  has no password by default

enord, 2022-10-31:
Thanks.   I thought that (as I hadn't used one initially).
Now gained access using "123456789" so problem solved.

closed 2022-10-31 by enord