s60sc/ESP32-CAM_MJPEG2SD #143: Corrupt IP address

enord, 2022-11-05:
I can see a WiFi network called:     ESP-CAM_MJPEG_D27EB994
When I attempt to connect to it, I am asked for a password.  
I can see in line 53 (from configs.txt copied below) that there is a password : 123456789
123456789 does not work and I cannot get access to the ESP-CAM_MJPEG_D27EB994 network.  

ST_ns2::0:Alt DNS server
AP_Pass:123456789:0:AP password
AP_ip::0:AP IP Address if not 192.168.4.1

I successfully used this "123456789" password once.

I can see in the Serial Monitor window that the cam is working however I am unable to view it via a web browser.   I connect to my wifi network (MyWiFi) and try to use IP address 192.168.2.222 from below but it will not connect.   I note there are commas after that address as shown in the last line below.  (192.168.2.222, , ,  )

[00:00:00.798 loadConfig] Load config
[00:00:00.881 updateAppStatus] Enabling motion detection
[00:00:00.940 remote_log_init_SD] Opened SD file for logging
[00:00:00.954 loadConfig] Using ssid: MyWiFi with static ip  192.168.2.222
[00:00:00.955 startWifi] Setting wifi hostname: ESP-CAM_MJPEG_D27EB994
[00:00:00.966 startWifi] Got stored router credentials. Connecting to: MyWiFi
[00:00:00.966 startWifi] Set config static ip: 192.168.2.222, , ,

s60sc, 2022-11-05:
Try blank password
For static IP also need to set gateway, subnet, dns
Passwords stored in flash. If get web access use Clear NVS button, otherwise erase flash from console

enord, 2022-11-05:
Thanks for replying
The computer wont accept a blank password so cant do that.
As I cant get web access, I cant use the Clear NVS button.

How do I erase flash from console?

s60sc, 2022-11-06:
Arduino IDE - Tools - Burn Bootloader

enord, 2022-11-07:
I have re-burnt the Bootloader - no issue.
Uploaded the program again from the IDE

It still attempts to connect to my router using the previously entered details which shows the PSRAM has not been erased.

As it cannot connect to my router, it starts the Access Point but I cannot connect as it asks for a password.   I have again tried 123456789 but no luck.

[00:00:00.472 setup] =============== Starting ===============
[00:00:00.543 infoSD] SD card type SDHC, Size: 15193MB, Used space: 3799MB, of total: 15185MB
[00:00:00.739 setup] Camera init OK
[00:00:00.784 loadConfig] Load config
[00:00:00.867 updateAppStatus] Enabling motion detection
[00:00:00.927 remote_log_init_SD] Opened SD file for logging
[00:00:00.941 loadConfig] Using ssid: MyWiFi with static ip  192.168.2.222
[00:00:00.941 startWifi] Setting wifi hostname: ESP-CAM_MJPEG_D27EB994
[00:00:00.952 startWifi] Got stored router credentials. Connecting to: MyWiFi
[00:00:00.952 startWifi] Set config static ip: 192.168.2.222, , , 
............................................................[00:00:31.062 startWifi] Unable to connect to router, start Access Point
[00:00:31.627 setWifiAP] Created Access Point with SSID: ESP-CAM_MJPEG_D27EB994
[00:00:31.628 setWifiAP] Use 'http://192.168.4.1' to connect
[00:00:31.664 setupMndsHost] mDNS service: http://ESP-CAM_MJPEG_D27EB994.local
[00:00:31.665 prepRecording] To record new AVI, do one of
[00:00:31.666 prepRecording] - move in front of camera

[00:00:31.685 startWebServer] Starting web server on port: 80
[00:00:31.687 startStreamServer] Starting streaming server on port: 81
[00:00:31.688 prepMic] Sound recording is Off
[00:00:31.698 prepSMTP] Email alerts not active
[00:00:31.699 prepUart] IO Extender not used
[00:00:31.705 setup] Camera Ready @ 20MHz, version 7.4
[00:00:31.706 checkMemory] Free: heap 106640, block: 77812, pSRAM 2550975
[00:00:40.986 processFrame] Capture started by Motion 
..................................................
..............................
[00:00:44.998 WARN closeAvi] Insufficient capture duration: 4 secs
[00:00:56.586 processFrame] Capture started by Motion 
....................
..................................................
....................................................................................................
..................................................
....................
[00:01:08.735 closeAvi] ******** AVI recording stats ********
[00:01:08.736 closeAvi] Recorded /19700101/19700101_000056_SVGA_20_12_240.avi
[00:01:08.746 closeAvi] AVI duration: 12 secs
[00:01:08.746 closeAvi] Number of frames: 240
[00:01:08.751 closeAvi] Required FPS: 20
[00:01:08.753 closeAvi] Actual FPS: 20.0
[00:01:08.753 closeAvi] File size: 3.38 MB
[00:01:08.764 closeAvi] Average frame length: 14770 bytes
[00:01:08.764 closeAvi] Average frame monitoring time: 2 ms
[00:01:08.775 closeAvi] Average frame buffering time: 0 ms
[00:01:08.780 closeAvi] Average frame storage time: 7 ms
[00:01:08.780 closeAvi] Average SD write speed: 1885 kB/s
[00:01:08.791 closeAvi] File open / completion times: 6 ms / 157 ms
[00:01:08.795 closeAvi] Busy: 21%
[00:01:08.795 checkMemory] Free: heap 114900, block: 77812, pSRAM 2132687
[00:01:08.810 closeAvi] *************************************
[00:01:08.810 checkFreeSpace] Card free space: 11382MB

closed 2022-12-19 by s60sc