s60sc/ESP32-CAM_MJPEG2SD #144: Reconnect after wifi down

maxbrax, 2022-11-06:
I pose the following question :
After the ESPCAM has connected to a wifi network, when the wifi network fails for some reason (e.g. due to a router power failure), the esp32 CAM after 30 seconds switches to access point mode and remains in access point mode even after the wifi signal has returned after a few minutes.

How can I modify the part of the script (utils.cpp) so that after a preset time the camera reconnects to the previously stored wifi network ?, could an automatic reset be performed ? after a preset time so that it automatically reconnects to the wifi network ?

s60sc, 2022-11-06:
Set Wifi config _Allow AP Fallback if no router_ to 0 if dont want AP

Else start a timer and then call doRestart() to connect to your previously stored wifi

rayray75063, 2023-01-04:
i have been trying to do that also.
i just cant get it to work correct.
any help and hints to making it work
is greatly appreciated

s60sc, 2023-01-04:
a new version will be uploaded shortly that allows simultaneous AP and STA

KamalW, 2023-01-08:
I am using ver: 8.1 and have "Allow simultaneous AP (0/1)" option value set to "0". But still the AP mode comes up. Serial monitor prints a line like this:

`16:44:00.007 -> [11:14:01.501 onWiFiEvent] Wifi AP SSID: ESP-CAM_MJPEG_xxxxxxxxxxxx started, use 'http://192.168.4.1' to connect`

For some reason the actual AP SSID is "ESP32_CAM" which is different from what is printed. Perhaps it is a name that I had set somewhere earlier. I can connect to it and it works as usual.

If the value of "Allow simultaneous AP (0/1)" option set to "1", the AP mode comes up as expected and uses the printed SSID.

s60sc, 2023-01-08:
seems esp32 starts up old stored APs automatically, will fix shortly

s60sc, 2023-01-08:
Version 8.2

KamalW, 2023-01-09:
Tested version 8.2 and now it is working as expected. Setting "Allow simultaneous AP (0/1)" option value to "0" does not start an AP now. If checked via serial monitor, output still indicates the start of the AP but also the immediate closure. Perhaps not starting at all is the best behavior but starting and closing immediately as experienced now is not bad.

closed 2023-01-15 by s60sc

deepak-coding-art, 2024-11-16:
What about the part where it should restart or retry to connect to WIFI after few minutes while the Allow simultaneous AP is 0 
how to do it ?