s60sc/ESP32-CAM_MJPEG2SD #148: SD card issue

chunter1, 2022-12-23:
I am using Arduino IDE 2.0.1 with Espressif ESP32 lib version 2.0.6 and get the following SDcard error on bootup:
I tried several ESP32 lib versions and SD-cards.
Does the SD-card have to be formated in a special way?
The ESP32 cam modul is this one here: https://www.reichelt.at/at/de/entwicklerboards-esp32-kamera-2mp-25--debo-cam-esp32-p266036.html?r=1
Any idea how to fix this?
Thanks!

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13836
load:0x40080400,len:3608
entry 0x400805f0

[00:02:46.844 setup] =============== Starting ===============
E (506) sdmmc_common: sdmmc_init_ocr: send_op_cond (1) returned 0x107
E (507) vfs_fat_sdmmc: sdmmc_card_init failed (0x107).
[00:02:46.872 ERROR @ utilsSD.cpp:79] SD card mount failed
[00:02:46.883 WARN startStorage] Insert SD card, will restart after 10 secs

s60sc, 2022-12-23:
Use the SDMMC_Test example sketch to check the reader and card (which needs to be FAT32 format)

chunter1, 2022-12-23:
The test with the SDMMC_Test example was successful.
I guess the flashing of the white LED during SD-card access is normal since it shares one of the GPIOs.
Any idea what else i could try?

chunter1, 2022-12-23:
Maybe a hint...
When using the SDMMC_Test example. the white LED flashes - however with the ESP32-CAM_MJPEG2SD code it doesn't.

chunter1, 2022-12-23:
In utilsSD.cpp i replaced
        res = SD_MMC.begin("/sdcard", true, sdFormatIfMountFailed);
with
        res = SD_MMC.begin();
and got at least a step further.

However now it tells me that it does not find a camera although the CameraWebServer example works.

============================

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13836
load:0x40080400,len:3608
entry 0x400805f0

[00:00:00.495 setup] =============== Starting ===============
[00:00:00.695 infoSD] SD card type SDHC, Size: 15193MB, Used space: 1MB, of total: 15020MB
E (691) camera: Camera probe failed with error 0x105(ESP_ERR_NOT_FOUND)
[00:00:00.708 ERROR @ ESP32-CAM_MJPEG2SD.ino:71] Camera init failed with error 0x105
E (702) gpio: gpio_set_level(226): GPIO output gpio_num error
E (803) gpio: gpio_set_level(226): GPIO output gpio_num error
E (803) gpio: gpio_install_isr_service(449): GPIO isr service already installed
E (816) camera: Camera probe failed with error 0x105(ESP_ERR_NOT_FOUND)
[00:00:00.833 ERROR @ ESP32-CAM_MJPEG2SD.ino:71] Camera init failed with error 0x105
E (827) gpio: gpio_set_level(226): GPIO output gpio_num error
E (927) gpio: gpio_set_level(226): GPIO output gpio_num error
ets Jun  8 2016 00:22:57

s60sc, 2022-12-23:
in globals.h do you have CAMERA_MODEL_AI_THINKER selected

chunter1, 2022-12-23:
With CAMERA_MODEL_AI_THINKER selected and "res = SD_MMC.begin("/sdcard", **false**, sdFormatIfMountFailed);" i now get to this point.
However, i can not connect to the webserver.
At least motion is already detected.

===============================

[00:00:00.938 loadConfig] Load config
[00:00:01.022 updateAppStatus] Enabling motion detection
[00:00:01.085 loadConfig] Using ssid: BLABLU  
[00:00:01.086 startWifi] Setting wifi hostname: ESP-CAM_MJPEG_3D82210
[00:00:01.086 startWifi] Got stored router credentials. Connecting to: BLABLU
[00:00:01.096 startWifi] Getting ip from dhcp ...
..[00:00:02.226 startPing] Started ping monitoring 
[00:00:02.226 startWifi] Use 'http://192.168.1.228' to connect
[00:00:02.227 prepRecording] To record new AVI, do one of:
[00:00:02.238 WARN getLocalNTP] Not yet synced with NTP
[00:00:02.238 prepRecording] - move in E (2223) sdmmc_cmd: sdmmc_read_sectors_dma: sdmmc_send_cmd returned 0xffffffff
E (2229) diskio_sdmmc: sdmmc_read_blocks failed (-1)
front of camera

[00:00:02.263 startWebServer] Starting web server on port: 80
[00:00:02.265 startStreamServer] Starting streaming server on port: 81
[00:00:02.265 prepMic] Sound recording is Off
[00:00:02.276 prepSMTP] Email alerts not active
[00:00:02.276 prepUart] IO Extender not used
[00:00:02.277 setup] Camera Ready @ 20MHz, version 7.4
[00:00:02.287 checkMemory] Free: heap 101640, block: 73716, pSRAM 2550243
E (6803) sdmmc_req: sdmmc_host_wait_for_event returned 0x107
E (6803) diskio_sdmmc: Check status failed (0x107)
E (29114) esp_jpg_decode: JPG Decompression Failed! Data format error

s60sc, 2022-12-23:
You still have SD card problems.
The SD card is used to store the web page which is why the web server wont work.

chunter1, 2022-12-23:
Got it running now!
I went back to the original code and replaced the 16GB with an 8GB SD-card.
Thank you very much for your support! :)

closed 2022-12-23 by chunter1