s60sc/ESP32-CAM_MJPEG2SD #149: I cannot get the onboard LED to work

dsfifty, 2023-01-06:
Using the default AI thinker board.
When I adjust the Lamp Slider, the serial window shows:
E (239600) ledc: ledc_update_duty(647): LEDC is not initialized
E (240519) ledc: ledc_set_duty(719): LEDC is not initialized
E (240519) ledc: ledc_update_duty(647): LEDC is not initialized
E (241646) ledc: ledc_set_duty(719): LEDC is not initialized
Am I doing something wrong?

dsfifty, 2023-01-06:
I think it's SD card related.
Stills don't record and time-lapse doesn't record.
I will try different cards.

s60sc, 2023-01-06:
Led - under Edit Config / Peripherals make sure lamp pin is 4 and use lamp is 1
Compile with arduino-esp32 v2.0.6

SD: stills are not stored only videos. Timelapse file only created at end of time lapse period

dsfifty, 2023-01-06:
Same SD, format with FAT, not FAT32 solved it.
Thanks for update about stills not being saved. 
As an alternative they can be saved from the browser using "Save Image As..."

closed 2023-01-06 by dsfifty