s60sc/ESP32-CAM_MJPEG2SD #152: Compilation issue: struct camera_config_t has no member

Dissssy, 2023-01-08:
Downloaded and attempted to flash the latest version to my new `AI Thinker ESP32-CAM` and am getting this compilation error

`
ESP32-CAM_MJPEG2SD:41:10: error: 'struct camera_config_t' has no member named 'pin_sccb_sda'; did you mean 'pin_sscb_sda'?
   config.pin_sccb_sda = SIOD_GPIO_NUM;
          ^~~~~~~~~~~~
          pin_sscb_sda
ESP32-CAM_MJPEG2SD:42:10: error: 'struct camera_config_t' has no member named 'pin_sccb_scl'; did you mean 'pin_sscb_scl'?
   config.pin_sccb_scl = SIOC_GPIO_NUM;
          ^~~~~~~~~~~~
          pin_sscb_scl
`

I changed the names as described there just in case there was a typo or something in the latest version, which compiled and uploaded, but then the sd card slot would not work

s60sc, 2023-01-08:
You need esp 2.0.6,

Dissssy, 2023-01-08:
i tried with esp 2.0.6 and was getting compilation errors but i'll give it another shot and get back to ya

Dissssy, 2023-01-08:
Alright so, i was running into missing header files issues and couldnt figure out where i was supposed to put them to make it work, so i tried using the arduino ide installer rather than getting it from the windows store, and after setting everything up again it worked first try with 2.0.6. now the only thing is i'm still getting Startup Failure: Check SD card inserted. 

do i need to format it to something specific? before i went through all this i tried using it with FAT32 but right now its unformatted

Dissssy, 2023-01-08:
tried it with FAT32 and unformatted and still getting the same error, no SD card inserted according to it

s60sc, 2023-01-08:
fat32. check errors on console log

Dissssy, 2023-01-08:
```
rst:0x1 (POWERON_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13836
load:0x40080400,len:3608
entry 0x400805f0
E (322) psram: PSRAM ID read error: 0xffffffff

[00:00:00.032 setup] =============== Starting ===============
E (378) sdmmc_common: sdmmc_init_ocr: send_op_cond (1) returned 0x107
E (378) vfs_fat_sdmmc: sdmmc_card_init failed (0x107).
[00:00:00.061 ERROR @ utilsFS.cpp:65] SD card mount failed
[00:00:00.071 loadConfig] Load config
[00:00:00.072 ERROR @ prefs.cpp:123] Failed to load file /data/configs.txt
[00:00:00.082 WARN retrieveConfigVal] Key hostName not set
[00:00:00.193 onWiFiEvent] Wifi Station started, connecting to: 
[00:00:00.194 onWiFiEvent] Wifi AP SSID:  started, use 'http://192.168.4.1' to connect
[00:00:00.204 onWiFiEvent] Wifi AP stopped: ESP_50D3AD
[00:00:00.205 onWiFiEvent] Wifi AP SSID: ESP_50D3AD started, use 'http://192.168.4.1' to connect
[00:00:00.215 onWiFiEvent] Wifi AP stopped: ESP_50D3AD
[00:00:00.215 WARN setWifiSTA] No Station SSID provided, use AP
[00:00:00.227 onWiFiEvent] Wifi AP SSID: ESP_50D3AD started, use 'http://192.168.4.1' to connect
[00:00:00.237 onWiFiEvent] Wifi AP stopped: ESP-CAM_MJPEG_ACD3509EF0C8
[00:00:00.238 onWiFiEvent] Wifi AP SSID: ESP-CAM_MJPEG_ACD3509EF0C8 started, use 'http://192.168.4.1' to connect
[00:00:06.464 startWebServer] Starting web server on port: 80
[00:00:06.464 ERROR @ ESP32-CAM_MJPEG2SD.ino:141] Startup Failure: Check SD card inserted
```

here's the full logs, i'm gonna try redownloading and flashing again, maybe i made a change somewhere i don't remember, it looks like there are a couple errors going on though so i'm not sure

s60sc, 2023-01-08:
E (378) vfs_fat_sdmmc: sdmmc_card_init failed (0x107).
not a software issue , esp hardware cant read card - try different one

Dissssy, 2023-01-08:
i guess i'll just have to use one that doesn't do saving to sd card then sadly, this is the only one i have lol, thanks for trying to help, i hope i can at least help some people figure out their compilation issues if anything

closed 2023-01-08 by Dissssy