s60sc/ESP32-CAM_MJPEG2SD #158: Access settings are not or only randomly loaded

chunter1, 2023-01-28:
When i change settings in the "access settings" menue and reboot the device, the settings are not shown after reboot finished.
Tried to reload the website and using firefox/chrome - still not working.
SD-card is a 8GB one but changing that also made no difference.

s60sc, 2023-01-28:
press the save button

chunter1, 2023-01-28:
Ofcourse i did press the save button before rebooting.

s60sc, 2023-01-28:
type http://<ip>/web?configs.txt and see if the file has changed

s60sc, 2023-01-28:
github auto corrected, should say "http://[ip]web?configs.txt"

s60sc, 2023-01-28:
github auto corrected again, should say "http://[ip]/web?configs.txt"

chunter1, 2023-01-28:
I guess we got a misunderstanding here.
It's not that the settings are ignored... they are simply not displayed on the webpage - no matter how often i press the browser reload button.
They only appear randomly.

[image]

s60sc, 2023-01-28:
implies a connection problem, open your browser inspector ctrl-shift-i to see what the network is doing

chunter1, 2023-01-29:
It's not a network issue since the other pages are working correctly.

closed 2023-01-30 by s60sc