s60sc/ESP32-CAM_MJPEG2SD #159: FTP not working

chunter1, 2023-01-28:
I set up an FPT server which is working fine with several other clients however with the ESP32-CAM_MJPEG2SD it isn't.
In the log of the FTP server there is nothing happening when the ESP32 cam is trying to upload files.
Here is the log of the ES32 cam:
[12:58:57.097 ERROR @ ftp.cpp:69] Error opening ftp connection to x.x.x.x:21
[12:58:57.098 ERROR @ ftp.cpp:171] Unable to make ftp connection

s60sc, 2023-01-28:
it cant see the server.  need a wifi conn to router

chunter1, 2023-01-28:
It does have a wifi connection - otherwise i could not ping the esp32 cam from my ftp server - which i can.

s60sc, 2023-01-28:
The error is reported by the espressif WiFiClient library so cant help

closed 2023-01-28 by s60sc