s60sc/ESP32-CAM_MJPEG2SD #164: No connection to Wi-Fi

Ihor59, 2023-02-01:
When i wrote after WIFI.begin

WiFi.begin(ST_SSID, ST_Pass);
    while( !WiFi.isConnected() ) {
    delay(1000);Serial.print(".");}

it connected to Wi-Fi but only without CD card
With CD card, it does not won't connect Wi-Fi

s60sc, 2023-02-01:
Cant really help with hardware problems
What does the log output

intek2016, 2023-02-18:
why i cant open my web browser 192,168.4.1,  i type my own ip but nothing change in serial monitor ..always use ip 192,168.4.1

intek2016, 2023-02-18:
[00:00:00.262 setup] =============== Starting ===============
[00:00:00.310 infoSD] SD card type SDHC, Size: 29824MB, Used space: 4MB, of total: 29812MB
[00:00:00.310 checkFreeSpace] Card free space: 29807MB
[00:00:00.521 prepCam] Camera init OK for model OV2640 on board CAMERA_MODEL_AI_THINKER
[00:00:00.568 loadConfig] Load config
[00:00:00.654 updateAppStatus] Enabling motion detection
[00:00:00.819 onWiFiEvent] Wifi Station started, connecting to: 
[00:00:00.819 onWiFiEvent] Wifi AP SSID: TP-LINK started, use 'http://192.168.4.1' to connect
[00:00:00.820 WARN setWifiSTA] No Station SSID provided, use AP 
[00:00:03.646 startWebServer] Starting web server on port: 80
[00:00:03.648 startStreamServer] Starting streaming server on port: 81
[00:00:07.648 WARN prepRecording] failed to get camera frame 
[00:00:07.649 prepRecording] To record new AVI, do one of:
[00:00:07.649 prepRecording] - move in front of camera

[00:00:07.659 setup] Camera model OV2640 on board CAMERA_MODEL_AI_THINKER ready @ 20MHz, version 8.4
[00:00:07.670 checkMemory] Free: heap 106828, block: 77812, pSRAM 450639

intek2016, 2023-02-18:
what the problem?
 plz help.. thx

closed 2023-03-11 by s60sc