s60sc/ESP32-CAM_MJPEG2SD #166: Cam dies after a few mins

galberti, 2023-02-04:
Everything works fine for a few mins, but after a while (no concrete time measurements done, ~5 mins) I get this in the log:

[WARN prepRecording] failed to get camera frame

Streaming, recording etc don't work anymore, Reboot ESP does not work, the only way to fix it is unplug and plug back power.
If I reboot I get the usual lines, here the relevant reported:
[..]
Camera init OK for model OV2640 on board CAMERA_MODEL_AI_THINKER
Camera model OV2640 on board CAMERA_MODEL_AI_THINKER ready @ 20MHz, version 8.4
Free: heap 99792, block: 69620, pSRAM 2547695
[..]

Any tip?

Thanks!

s60sc, 2023-02-04:
some ov2640 modules wont work at 20Mhz, try 10.

galberti, 2023-02-04:
I am afraid it's the overheating, still unsure if the cam or other. Thanks anyway, and great job on the sketch!

closed 2023-02-04 by galberti