s60sc/ESP32-CAM_MJPEG2SD #169: Can only connect to AP

awskmet, 2023-02-09:
Thanks so much for the code, I think it's awesome. This is the first arduino-type project I've ever tried, I'm a Java developer, My issue is that I can only connect to the camera if I select ESP-CAM_MJPEG_xxxx  nd navigate to 192.168.4.1. If I try that IP adress when I am connected to my router it times out (resource not found). Is it possible to connect  to the camera (see the main page with start stream/etc) from my laptop while my laptop is connected to my router instead of the camera's access point? I can interact with it from my code much more easily this way

s60sc, 2023-02-09:
Thanks
You need to enter router SSID and password in settings

awskmet, 2023-02-10:
Thanks for the quick response. You are the man. I had those settings enabled but did not realize the ip adress to connect to was different than the one I was connecting to during AP mode. Thanks again

closed 2023-02-10 by awskmet