s60sc/ESP32-CAM_MJPEG2SD #170: MQTT Support

mkarliner, 2023-02-10:
Would you consider including MQTT support?

That would be outgoing messages to notify of motion detections,
captures etc, and incoming instructions to say start or stop recording?

That way node-red and other flows can monitor and control the camera.

s60sc, 2023-02-11:
At some point when I have time I will look at mqtt / telegram / discord integration.

gemi254, 2023-03-14:
I've made an integration for ESP32-CAM_MJPEG2SD..
Check this out 
https://github.com/gemi254/ESP32-CAM_MJPEG2SD/tree/ESP32-CAM_MJPEG2SD

Upload an navigate to camera interface Edit config -> Others 
Edit the mqtt_broker_ip, post, pass etc.
Mqtt will auto connect if configuration is not blank on ping success.
It will send messages Record On/Off Motion On/Off to the mqtt broker on channel /status.
topic: homeassistant/sensor/ESP-CAM_MJPEG_904CAAF23A08/status -> {"MOTION":"ON", "TIME":"10:07:47.560"}

You can also publish control commands to the /cmd channel in order to control camera.
topic: homeassistant/sensor/ESP-CAM_MJPEG_904CAAF23A08/cmd ->  dbgVerbose=1;framesize=7;fps=1

closed 2023-04-16 by s60sc