s60sc/ESP32-CAM_MJPEG2SD #172: can't open connect to camera

intek2016, 2023-02-18:
i dont get ip that type on code. always get to use http:192.168.4.1
but 192.168.4.1 cant open di web browser too
what the problem ?

this my serial monitor when im reset esp32 cam :

ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13836
load:0x40080400,len:3608
entry 0x400805f0

[00:00:00.259 setup] =============== Starting ===============
[00:00:00.305 infoSD] SD card type SDHC, Size: 29824MB, Used space: 18MB, of total: 29812MB
[00:00:00.306 checkFreeSpace] Card free space: 29793MB
[00:00:00.511 prepCam] Camera init OK for model OV2640 on board CAMERA_MODEL_AI_THINKER
[00:00:00.557 loadConfig] Load config
[00:00:00.643 updateAppStatus] Enabling motion detection
[00:00:00.806 onWiFiEvent] Wifi Station started, connecting to: 
[00:00:00.806 onWiFiEvent] Wifi AP SSID: ESP-CAM_MJPEG_44F86B1B5AE0 started, use 'http://192.168.4.1' to connect
[00:00:00.807 WARN setWifiSTA] No Station SSID provided, use AP 
[00:00:03.833 startWebServer] Starting web server on port: 80
[00:00:03.834 startStreamServer] Starting streaming server on port: 81
[00:00:07.834 WARN prepRecording] failed to get camera frame 
[00:00:07.835 prepRecording] To record new AVI, do one of:
[00:00:07.835 prepRecording] - move in front of camera

[00:00:07.846 setup] Camera model OV2640 on board CAMERA_MODEL_AI_THINKER ready @ 20MHz, version 8.4
[00:00:07.856 checkMemory] Free: heap 106764, block: 77812, pSRAM 450607

intek2016, 2023-02-18:
im connet to SSiD ESP-CAM_MJPEG_44F86B1B5AE0 have  error:

[00:00:58.657 ERROR @ webServer.cpp:53] File does not exist or cannot be opened: /data/MJPEG2SD.htm

s60sc, 2023-02-18:
Use OTA to download files from your pc, or connect to router and it will download automatically

intek2016, 2023-02-18:
I have error when open 192.168.4.1 in web browser : 

[00:00:58.657 ERROR @ webServer.cpp:53] File does not exist or cannot be opened: /data/MJPEG2SD.htm

What that mean?  Sorry my english not good.

intek2016, 2023-02-18:
im allready upload code to esp32 cam and reset the esp32. but cant open the cam menu in my web browser

s60sc, 2023-02-18:
copy the /data folder and its contents to your sd card

intek2016, 2023-02-19:
how do i enable OTA to auto download from github /data .

the current condition is still in the web browser saying "File does not exist or cannot be opened"

I have bought an online sdcard reader via online. but now still waiting for the item to arrive. the plan is to copy the data folder manually

s60sc, 2023-02-19:
what does the arduino monitor log say?

intek2016, 2023-02-19:
not automatically downloaded from GitHub on app startup

[00:00:02.106 setup] =============== Starting =============
[00:00:02.153 infoSD] SD card type SDHC, Size: 29824MB, Used space: 25MB, of total: 29812MB
[00:00:02.154 checkFreeSpace] Card free space: 29786MB
[00:00:02.359 prepCam] Camera init OK for model OV2640 on board CAMERA_MODEL_AI_THINKER
[00:00:02.405 loadConfig] Load config
[00:00:02.490 updateAppStatus] Enabling motion detection
[00:00:02.780 onWiFiEvent] Wifi Station started, connecting to: 
[00:00:02.781 onWiFiEvent] Wifi AP SSID: ESP32CAMPROJECT started, use 'http://192.168.4.1' to connect
[00:00:02.781 WARN setWifiSTA] No Station SSID provided, use AP 
[00:00:05.607 startWebServer] Starting web server on port: 80
[00:00:05.609 startStreamServer] Starting streaming server on port: 81
[00:00:05.610 prepRecording] To record new AVI, do one of:
[00:00:05.620 prepRecording] - move in front of camera

[00:00:05.621 setup] Camera model OV2640 on board CAMERA_MODEL_AI_THINKER ready @ 20MHz, version 8.4
[00:00:05.631 checkMemory] Free: heap 106832, block: 77812, pSRAM 450639
[00:05:16.336 onWiFiEvent] WiFi AP client connection
[00:05:21.069 ERROR @ webServer.cpp:53] File does not exist or cannot be opened: /data/MJPEG2SD.htm

s60sc, 2023-02-19:
[00:00:02.781 WARN setWifiSTA] No Station SSID provided, use AP

You dont have an internet connection.

intek2016, 2023-02-20:
i have  internet this my code in utils.cpp
 something wrong ?

/************************** Wifi **************************/

char hostName[32] = ""; // Default Host name
char ST_SSID[32]  = "TP-LINK"; //Default router ssid
char ST_Pass[MAX_PWD_LEN] = "intek2016"; //Default router passd

// leave following blank for dhcp
char ST_ip[16]  = ""; // Static IP
char ST_sn[16]  = ""; // subnet normally 255.255.255.0
char ST_gw[16]  = ""; // gateway to internet, normally router IP
char ST_ns1[16] = ""; // DNS Server, can be router IP (needed for SNTP)
char ST_ns2[16] = ""; // alternative DNS Server, can be blank

// Access point Config Portal SSID and Pass
char AP_SSID[32] = "ESP32CAMPROJECT";
char AP_Pass[MAX_PWD_LEN] = "";
char AP_ip[16]  = ""; //Leave blank to use 192.168.4.1
char AP_sn[16]  = "";
char AP_gw[16]  = "";

// basic HTTP Authentication access to web page
char Auth_Name[16] = ""; 
char Auth_Pass[MAX_PWD_LEN] = "";

intek2016, 2023-02-20:
this serial monitor when i have connected to router but have error : Check SD card inserted 

[00:00:02.052 setup] =============== Starting ===============
E (366) sdmmc_sd: sdmmc_init_sd_scr: send_scr (1) returned 0x107
E (367) vfs_fat_sdmmc: sdmmc_card_init failed (0x107).
[00:00:02.159 ERROR @ utilsFS.cpp:65] SD card mount failed 
[00:00:02.365 prepCam] Camera init OK for model OV2640 on board CAMERA_MODEL_AI_THINKER
[00:00:02.412 loadConfig] Load config
[00:00:02.412 ERROR @ prefs.cpp:125] Failed to load file /data/configs.txt 
[00:00:02.647 onWiFiEvent] Wifi Station started, connecting to: TP-LINK
[00:00:02.648 setWifiSTA] Wifi Station IP from DHCP
[00:00:02.648 onWiFiEvent] Wifi AP SSID: ESP32CAMPROJECT started, use 'http://192.168.4.1' to connect
[00:00:02.713 onWiFiEvent] WiFi Station connection to TP-LINK, using hostname: 
[00:00:02.974 onWiFiEvent] Wifi Station IP, use 'http://192.168.0.101' to connect
[00:00:03.162 startPing] Started ping monitoring
[00:00:03.883 onWiFiEvent] WiFi AP client connection
[00:00:05.977 startWifi] Wifi stats for TP-LINK - signal strength: -55 dBm; Encryption: WPA_WPA2_PSK; channel: 11
[00:00:05.979 startWebServer] Starting web server on port: 80
[00:00:05.990 ERROR @ ESP32-CAM_MJPEG2SD.ino:147] Startup Failure: Check SD card inserted 
[00:00:06.417 getLocalNTP] Using NTP server: pool.ntp.org
[00:00:06.417 WARN getLocalNTP] Not yet synced with NTP 
[00:00:06.418 ERROR @ setupAssist.cpp:90] Open failed: /data/configs.txt 
[02:39:25.244 getLocalNTP] Using NTP server: pool.ntp.org
[02:39:25.244 showLocalTime] Got current time from NTP: 20/02/2023 02:39:25 with tz: GMT0
[02:39:25.255 ERROR @ setupAssist.cpp:90] Open failed: /data/configs.txt 
[02:39:55.246 ERROR @ setupAssist.cpp:90] Open failed: /data/configs.txt 
[02:40:25.254 ERROR @ setupAssist.cpp:90] Open failed: /data/configs.txt 
[02:40:55.248 ERROR @ setupAssist.cpp:90] Open failed: /data/configs.txt

s60sc, 2023-02-20:
E (366) sdmmc_sd: sdmmc_init_sd_scr: send_scr (1) returned 0x107
E (367) vfs_fat_sdmmc: sdmmc_card_init failed (0x107).
[00:00:02.159 ERROR @ utilsFS.cpp:65] SD card mount failed

Your SD card can't be read. You need to check and fix errors

intek2016, 2023-02-20:
it seems my esp32 cam is not stable. because it always gets a definite error. I will try with a new one. to compare them.

intek2016, 2023-02-20:
now my esp32 camera is running pretty well. but I have to try and learn more about the existing settings. I see there is an option to activate the flash with the PIR sensor. is there an option to activate the light only with motion if it is detected without using PIR?

closed 2023-03-11 by s60sc