s60sc/ESP32-CAM_MJPEG2SD #175: How to access Camera Webserver from outer networks?

maximagination1, 2023-02-24:
My current camera works based on local IPs it's assigned to that change from time to time, but what would I need to do so I can access the web server from any other network? Could it be through port forwarding? DDNS? 

Also, how would I go about making those changes?

Thanks.

s60sc, 2023-02-24:
Port forward port numbers 80, 81, 82 for camera ip

maximagination1, 2023-02-25:
Ok, thanks for that.

After Port Forwarding the IP, do you know what I would enter in my web browser from another network to access the web server? Entering the same local IP I port forwarded wouldn't make sense as I'd be in another network.

s60sc, 2023-02-25:
The [external ip](https://whatismyipaddress.com/) of the router that the camera is connected to

maximagination1, 2023-02-25:
Oh ok. Thanks for your help.

maximagination1, 2023-02-26:
Hi again. 

After some attempts, I still couldn't access my ESP32-CAM remotely using the external IP of my router even after port forwarding the camera's IP with those port numbers. I've been stuck for a while...

Could you please help with my misunderstanding?

nnjim, 2023-02-26:
Presumably you have set a port forward rule in your router for the camera's IP address and the ports as mentioned above.

Be aware though that you can only browse to the camera via the external IP if that browsing device is not connected to the internal network. EG if you are using a mobile phone connected via a mobile data network, you must switch off its wifi.

s60sc, 2023-02-26:
Your router should have entries like:

[image]

Your external ports must be unused ports, dont use 80 -82 as external ports or all your web traffic will be directed to your camera. Also note comment from @nnjim 

Enter eg external_ip:8100 to access camera

closed 2023-03-11 by s60sc