s60sc/ESP32-CAM_MJPEG2SD #177: PORT

Makss8, 2023-03-03:
Hot to change straming port from 81 to 4138

s60sc, 2023-03-04:
in `streamServer.cpp` change:

```
  config.server_port += 1;
  config.ctrl_port += 1;
```

s60sc, 2023-03-04:
default web server port = 80
stream server port = default + 1 = 81
control port not used

`config.server_port = 4138;`

s60sc, 2023-03-04:
See Readme

Compile with PSRAM enabled and the following Partition scheme:

ESP32 - Minimal SPIFFS (...)
ESP32S3 - 8M with spiffs (...)

Makss8, 2023-03-04:
[image]
I changed port and now i dont see image stream: 
[image]
[image]

s60sc, 2023-03-04:
you need to edit `mjpeg2sd.htm`

`        const streamUrl = baseHost + ':81';`

Makss8, 2023-03-04:
Thank you sooo soooo much for your help! It works!!!!! If i want change webserver port i must change something here? static httpd_handle_t httpServer = NULL; // web server listens on port 80

s60sc, 2023-03-04:
Add `config.server_port = n;` to `startWebServer()` to override default

Makss8, 2023-03-04:
Why is the camera doing AP even though I turned off the Allow simultaneous AP option? Additionally, the AP is called ESP_76B155 and not Cam 
[image]

s60sc, 2023-03-04:
Press `Save `to make changes persistent

Makss8, 2023-03-04:
I did the save a lot of times and restart, at first everything was ok and then this AP point was done. Even on the card in the config file there are data with the name of the AP point and it does the default name anyway

MisiekDP, 2023-03-11:
I have totally this same problem with AP like @Makss8. I love this project but there is some errors too.
Problem with AP:
[image]
[image]

Btw. there is a problem with entered password - the AP is all the time unprotected and this is very unsafe for example in my situation.
The one way to secure is only to add authentication settings:
[image]
But is is not the best option for me and not works in all situations (for example I set login "a" and password "a", clicked "Save" and I blocked my ESP - not possible to get access, but with "admin" and "admin" it works).

s60sc, 2023-03-11:
New version 8.5

closed 2023-03-11 by s60sc

rbolero, 2023-04-16:
> Add `config.server_port = n;` to `startWebServer()` to override default

where can I find this startwebserver? is it on .ino file?

s60sc, 2023-04-16:
In `appGlobals.h` change:

```
// web server ports
#define WEB_PORT 80 // app control
#define STREAM_PORT 81 // camera images
#define OTA_PORT 82 // OTA update
```

In` mjpeg2sd.htm` change:
```
const webPort = 80;
const otaPort = 82;
const streamPort = 81;
```