s60sc/ESP32-CAM_MJPEG2SD #178: FTP connection failed

lztz0022, 2023-03-10:
Sir s60sc, thank you for your wonderful works and generous sharing.

With your project, I can easily connect to ESP32cam and run it successfully, but the FTP automatic upload function has encountered problems, regardless of whether I fill in the Ftp Server with "ftp://ipaddress" Or directly fill in "ipaddress". The serial port monitor returns:
```

ERROR @ ftp.cpp:57] Command got wrong response: SSH -2.0-OpenSSH_ 7.6p1 Ubuntu-4ubuntu0.3

ERROR @ ftp.cpp:171] Unable to make ftp connection
```
I tried to connect to the FTP server with WinSCP, and I can successfully connect server and upload or download data.
Please, could u help me?

s60sc, 2023-03-10:
Looks like server requires sftp, cam only works with ftp

lztz0022, 2023-03-10:
Thank you for your quick reply. I will check the server settings.
And there is a function that is not urgently needed, but can be considered, just like wireless router, which can hide the SSID of AP mode.

s60sc, 2023-03-10:
In `utils.cpp`, try replacing:
`WiFi.softAP(AP_SSID, AP_Pass);`
with:
`WiFi.softAP(AP_SSID, AP_Pass, 1, 1);`

closed 2023-03-11 by s60sc

lztz0022, 2023-03-11:
I apologize for the interruption caused by reopening the issue, but I think every reply to the request deserves a formal response. Thanks to your help, now my esp32 camera is running pretty well. Love your awesome project.