s60sc/ESP32-CAM_MJPEG2SD #179: Small issues and ideas

MisiekDP, 2023-03-11:
Hey @s60sc,
I love your project, it is awesome but I see a small issues and maybe small ideas.

**1. Recorded file is not auto uploaded when one file is still at uploading progress**
For example, I have checked Auto upload function with FTP and it works perfect but when I record video and it is still in upload progress and I record a new video (the new video is saved when previous file is still uploading), I got an error that file can not be uploaded. And in general way, not all files are uploaded to FTP. Maybe could correct it to add something like auto upload queue for new recorded videos too?

**2. Problem with changing AP settings (name, password, allowing simultaneous AP)**
I described it here: [LINK](https://github.com/s60sc/ESP32-CAM_MJPEG2SD/issues/177#issuecomment-1464862120)

**3. Any description for actions mode on SD min free**
I can't find any documentation where 0, 1 or 2 method is explained. It could be very helpful which option is the best for user.
[image]

And I thank you very much for this project, it is really awesome. I would like to use it very much 👌😉

s60sc, 2023-03-11:
Thanks

1. Good point but its not something I want to spend time implementing. Anyone else is welcome to contribute.

2. If an AP was previously stored before the app was installed, ESP automatically starts it first.
I'm working on a fix but in the meantime try completely erasing the flash and reinstalling

3 In `utilsFS.cpp`: 

```
// Storage settings
int sdMinCardFreeSpace = 100; // Minimum amount of card free Megabytes before sdFreeSpaceMode action is enabled
int sdFreeSpaceMode = 1; // 0 - No Check, 1 - Delete oldest dir, 2 - Upload to ftp and then delete folder on SD 
bool formatIfMountFailed = true; // Auto format the file system if mount failed. Set to false to not auto format.
```

s60sc, 2023-03-11:
New version 8.5

closed 2023-03-11 by s60sc

MisiekDP, 2023-03-15:
> Thanks
> 
> 1. Good point but its not something I want to spend time implementing. Anyone else is welcome to contribute.

Ok, If I could know about programming on ESP32 I could help but now I am still with basics on ESP8266 :)

> 2. If an AP was previously stored before the app was installed, ESP automatically starts it first.
>    I'm working on a fix but in the meantime try completely erasing the flash and reinstalling

Ok, I did this and yea, on version 8.4 erasing flash was helpful. And on 8.5 the AP password works nice (only with password like 'qazxsw987Dx' but with 'qwerty' I have a problem (the result was default unprotected AP like 'ESP_XXX')).
Now I see that I have a problem with changing a SSID on AP. Generally I cant change it after flash via Arduino IDE.
My example: I have 'ESP-CAM_MJPEG_1' SSID and when I change to 'ESP-CAM_MJPEG_AP_1' I have still the old SSID.

> 3 In `utilsFS.cpp`:
>
> ```
> // Storage settings
> int sdMinCardFreeSpace = 100; // Minimum amount of card free Megabytes before sdFreeSpaceMode action is enabled
> int sdFreeSpaceMode = 1; // 0 - No Check, [...]

Thanks :)

s60sc, 2023-03-15:
_but with 'qwerty' I have a problem_ 

ESP AP library requires a minimum password length of 8

MisiekDP, 2023-03-15:
> _but with 'qwerty' I have a problem_
> 
> ESP AP library requires a minimum password length of 8

Ok, thanks and what about the problem of changing the SSID of AP?
@s60sc

s60sc, 2023-03-15:
Fix applied