s60sc/ESP32-CAM_MJPEG2SD #182: Cannot view with Safari on Apple devices

YapCK45, 2023-03-15:
I was viewing the web cam (v8.5) using my PC and it works very well. I tried with my iPhone and could not see the video. Screen shot attached
[image]

Did not have this issue with previous versions (v7.4)

s60sc, 2023-03-15:
try scrolling down, or use chrome
I dont have access to an apple device so cant check

YapCK45, 2023-03-15:
I have tried chrome besides safari and also scrolling down. Cannot see the video. Tried both on iPhone and iPad. Hopefully someone using apple devices has the solution.

YapCK45, 2023-03-20:
Tried Opera and Firefox and also cannot view. Would appreciate any advice from any one using Apple devices. Thank you.

than-sad, 2023-03-23:
Hi, I have the same problem, in my case I can use it with chrome, but depending of the quality I setup or maybe randomly camera stops working.

I also checked the safari inspector due to I don't see the camera frame (canvas) and I can see this errors:

`Unhandled Promise Rejection: TypeError: document.exitFullscreen is not a function. (In 'document.exitFullscreen()', 'document.exitFullscreen' is undefined)`

Line 2261:
`if ((selectedSize == 'actualSize' || selectedSize == 'maxFit') && document.fullscreenElement !== null) document.exitFullscreen(); `

I think it's working in safari but can't be displayed due to this error. Because I can on and off the flash light.

Also is doing strange behavior, sometimes trying to reload the page I can see the code in the browser (chrome and safari).

s60sc, 2023-03-23:
Interesting, seems safari doesnt support `exitFullscreen`

Try replacing `document.exitFullscreen()` with:
```
           {
              if (document.exitFullscreen) document.exitFullscreen();
              else document.webkitExitFullscreen(); 
            }
```

and each instance of `viewContainer.requestFullscreen()` with:
```
     if (viewContainer.requestFullscreen) viewContainer.requestFullscreen();
     else viewContainer.webkitRequestFullscreen();
```
Cant test it myself, may also be other differences.

than-sad, 2023-03-24:
Hi! Effectively adding webkit make it work in safari on Mac, not on iOS but I read on internet exitfullscreen isn't available in iOS. I'll try to see how its made the original example from expressive because with that version works properly.

Thanks for your fast answer and help and great job with this project.

YapCK45, 2023-03-25:
I tried viewing in on an Android device using Chrome with similar results - unable to view streaming video. Even when setting Chrome to show in desktop mode.
Just a note: seems like MaxImagination could view video on a smartphone. If I remember correctly, this was shown on his Youtube video. Wonder how he managed to do it.

than-sad, 2023-03-25:
Hi!

I tried with an old iPad mini and works properly but not in iPhone 14 last iOS that’s confusing me why with one works and not with the other one…
@s60sc I found this reference from Apple developers but my acknowledgment isn’t enough to understand with isn’t working, maybe you will: https://developer.apple.com/documentation/webkitjs/htmlvideoelement/1628805-webkitsupportsfullscreen

Regards, Than.

YapCK45, 2023-03-25:
@than-sad I tried with an old iPad (really old one) but it didn't work. This is most puzzling.

than-sad, 2023-03-25:
> @than-sad I tried with an old iPad (really old one) but it didn't work. This is most puzzling.

Hi! Have you modified the code as @s60sc suggested? Just open the html from SD root folder and change the lines. After that it should work on iPad and safari on MacOS. Not in iPhone. I’m reading on internet people are claiming Apple to give support for that from years ago. Someone say Apple don’t want to allow full screen on iPhone to don’t lost sales in video games on AppStore due to allow people play games on full screen in iPhones.

s60sc, 2023-03-25:
GPT4 says:

> Yes, iOS allows apps and videos to run in fullscreen mode on iPhones. For instance, when watching a video using the native video player or within apps like YouTube, you can enter fullscreen mode by **rotating your iPhone to landscape orientation**

than-sad, 2023-03-25:
> GPT4 says:
> 
> > Yes, iOS allows apps and videos to run in fullscreen mode on iPhones. For instance, when watching a video using the native video player or within apps like YouTube, you can enter fullscreen mode by **rotating your iPhone to landscape orientation**

[image]

Maybe the way Fullscreen is called isn’t the proper one. I don’t know, I’m designer not programmer 😔

s60sc, 2023-03-25:
GPT4 lied 😲

than-sad, 2023-03-27:
Hi! @YapCK45 to make it work on iPhone (Removing anterior webkit additions), you just need to comment three lines (2251, 2257 & 2261) with // as follows in /data/MJPEG2SD.html:
[image]

[image]
Still need some adjustments but at least is operational. Regards.

than-sad, 2023-03-27:
Hi! I checked and inspect element doesn’t show any error and tried too with safari in Mac and isn’t working too (before comment that lines was working).

I don’t know the reason, maybe @s60sc can help us realise why commenting that lines player stoped working.

Regards.

s60sc, 2023-03-28:
sorry, unable to help with apple issues

YapCK45, 2023-04-01:
Good detective work @than-sad . Have modified the code as suggested and it works! Thanks for confirming that it **cannot** do playback on iPhone. Hopefully someone else using Apple devices can find out more fixes.

Noticed that when viewed on iPhone, the resolution had several **new** options - QSXGA (2560x1920) down to FHD (1920x1080). This option does not appear when viewing on the desktop. Perhaps @s60sc can comment on that.

YapCK45, 2023-04-01:
Want to put in another observation. There seems to be a noticeable lag for the video to appear when the **start stream** button is clicked. Usually not very obvious. But sometimes, it can take several seconds. I thought the system had hung, but it's not because the uptime is still running.
Will do some more investigation to check if it is due to the modification by comparing with the original.

than-sad, 2023-04-01:
Hi, in safari for MacOS also shows this extra resolutions, I think its due to don't restrict by camera model because all this extra resolutions doesn't work with OV2640. I bought a ESP32-S3 as @s60sc recommends with OV3660. I will share the differences when I receive it.

The playback isn't working too in safari for MacOS with the original HTM (only works in Chrome). I hope some MacOS/iOS user will give us some help.

The delay, also happens to me some times but I think is related to no responsive camera, it rarely happens to me, I just refresh the page or reboot the device to solve it.

Regards.

YapCK45, 2023-04-01:
I did some research and there are 3 camera modules that can be used with the ESP32Cam - OV2640 (2 megapixels), OV3660 (3 megapixels) and OV5640 (5 megapixels). 
Overall, the OV5640 has the highest resolution and additional features (auto-focus, auto WB), while the OV3660 has better low-light performance than the other two sensors. The OV2640 has a smaller form factor and lower power consumption, making it suitable for low-power applications.
The cost difference of OV3660 and OV5640 is marginal; so I am thinking of getting the latter. My concern is whether the ESP32Cam can handle the higher resolution.
@s60sc Would you be able to comment?

than-sad, 2023-04-01:
@YapCK45 you can see it in the project readme. @s60sc mention the OV3660 and OV5640. I bough the OV3660 fisheye after read it.

Regards.

YapCK45, 2023-04-01:
Thanks for the heads up. Amazed that @s60sc thought of so many things. Great job!
Will check out v8.61 later.

s60sc, 2023-04-01:
I dont have an OV3660 so will be interested in how it does.

An OV5640 is best used with an ESP32S3, but mount it on a heatsink. The autofocus needs a separate driver, but autofocus is only needed for close up.

I use this [ESP32S3](https://github.com/Freenove/Freenove_ESP32_S3_WROOM_Board) bought from aliexpress, but throw away the included SD card. The included cam is an OV2640 though.

than-sad, 2023-04-01:
I bought the ESP32S3 (waiting for arrival) that you lined to AliExpress (it's the only one available) and an extra OV3660 fish eye to put it on the house entrance. I will give you feedback.

closed 2023-07-09 by s60sc