s60sc/ESP32-CAM_MJPEG2SD #187: finding ip problem

DJprasenjit, 2023-03-18:
every time i restart camera and router ip changes. i have lot more esp devices. having problem to find my ip.
can it be possible to show the ip to telegram bot .
the red led is no use. please use it as status led. like recording blink at frames. or saving to sd card red led on. or any error blink 1sec on off
time laps is recoding is not understandable. when it reboots then time laps starts but cant understand its working or not. flash white led on frame. and add some option to manual time laps video record and setting on web server.
thanks a lot for making this good project

s60sc, 2023-03-18:
Define a static IP in WiFi config

DJprasenjit, 2023-03-19:
ok sir.
the other issues please check.
i cant understand the time laps. is it working or not?
and the gmail smtp server? i already set id pass of smtp but nothing.

s60sc, 2023-03-19:
Ask ChatGPT

Explain time lapse feature of esp32_mjpeg2sd app

The ESP32-MJPEG2SD app is a software application designed to capture video from an MJPEG (Motion JPEG) camera and store it on an SD card. The app includes a time-lapse feature that allows you to capture a series of images at specified intervals and then compile them into a video.

To use the time-lapse feature, you first need to set the capture mode to time-lapse in the app settings. Then, you can configure the time-lapse interval, which determines how often the app captures an image. For example, if you set the interval to 10 seconds, the app will capture an image every 10 seconds.

Once you have configured the time-lapse settings, the app will capture a series of images at the specified intervals and store them on the SD card. You can then use video editing software to compile the images into a time-lapse video.

The time-lapse feature is useful for capturing slow processes or changes that occur over a long period of time, such as the growth of a plant, the movement of clouds, or the construction of a building. It allows you to compress hours or days of footage into a short video, making it easier to visualize and analyze the changes over time.

atesis, 2023-03-24:
hello. I have a problem. I used this video [https://www.youtube.com/watch?v=k_PJLkfqDuI&t=1s ] as the main guide when using your modules and used all the port settings as in that video, I tried to flash using 3 different programmers including ESP32-CAM-MB
[https://sun9-6.userapi.com/impg/bP0stfmXiJXN5_cGIg1AulJgvGihyQhzD6n7ag/TgvvJDDfuXY.jpg?size=1215x2160&quality=95&sign=1f9290fa0dadcb88052df7635b486b4e&type=album](url)
[https://sun9-40.userapi.com/impg/ZW_hPlaLya7lcmqmpk6hMP49XM6lBV8tHd8_3A/QRYHf6gvmqQ.jpg?size=1215x2160&quality=95&sign=db82f07a7aeddd361ef6d88c447c8902&type=album](url)
[https://sun9-41.userapi.com/impg/hmDHwk6ohfa-ARcI3bL1ylSLopX6Ada0TnsTRg/52TGJp9hEEo.jpg?size=1215x2160&quality=95&sign=5e76ac51869dad72d3e89a9b487b8181&type=album](url)
[https://sun9-62.userapi.com/impg/rlrwn_dXutpPVJSvT14kbz9ndz3-pZce7t4MBA/eONT-ktRZi4.jpg?size=1215x2160&quality=95&sign=d1f67abb177ce74d7a2bd856e566b089&type=album](url)

When sketch was loaded i took the message "Hard resetting via RST Pin"[https://sun9-49.userapi.com/impg/aqPOr00gmctmafoRTwQxPv83Wffsr7NlLjKSJA/Q_a978pfe4o.jpg?size=898x564&quality=96&sign=8f3f903baca9ff744111e9d89bcd9cf6&type=album](url)

Then after disconnecting the jumper from IOO to GND, I pressed RST on the board, but each time I did not get an ip, instead the port monitor starts displaying some characters in an incomprehensible encoding
[https://sun9-69.userapi.com/impg/uY3HsZ-J6QMtp-5QjHONGHT3fde_IlA0GG9qKA/IbEj_rWO_B8.jpg?size=895x559&quality=96&sign=4c2efee985217fc5888b329f43d639cb&type=album](url)
do you know what the problem is?
thank you for your attention!

s60sc, 2023-03-24:
set baud rate to 115200

DJprasenjit, 2023-03-25:
if you did everything good then it should be on. but in your case ithing its rebooting . in this case you need a strong power supply 5v to power esp32 cam board. use battry or boost converter 5v

DJprasenjit, 2023-03-25:
mdns name services in order to use http://[Host Name] instead of ip address.
 please tell me this function works or not. i am having issue to find ip.

atesis, 2023-03-26:
> if you did everything good then it should be on. but in your case ithing its rebooting . in this case you need a strong power supply 5v to power esp32 cam board. use battry or boost converter 5v

I tried it, thanks for the comment.

> set baud rate to 115200
I changed the baud rate to 115200, it really helped, now the port has started to output messages.

now I apparently have a problem with Wi-Fi, I tried to connect to a working Wi-Fi, but nothing worked out

he also started to give an error that there is no SD card - this is critical for the esp32 cam to transmit the stream to the specified address, because I don't need to store these images on the sd card, I just wanted to use it as an output image in the form of a stream and save the images on another device
[image]

I thought that the problem is in a working Wi-Fi, and it's worth trying at home

[image]

again, nothing happened, among other wifi connection errors, he gave me 2 IP addresses. when connected to one (192.168.4.1), the site simply does not open, and when connected via the link 192.168.1.76, he writes to me that there is no SD card.

[image]

I was looking for the connection scheme according to which I programmed the device and came across the fact that in the one on the circuit the jumper on it stands at 5 volts, although the videos on youtube say that the jumper should stand at 3 volts, in some circuits there is no jumper at all and power is supplied only to the 3.3 volt pin. in general, I haven't tried switching the jumper yet, I'm afraid to burn the camera, since I don't quite understand where these 5 volts will go from the programmer.

[image]

I also tried powering the esp32 cam separately from the programmer's power supply using a 5 volt 400 mA power supply as shown in the diagram below
[image]

again, the wifi connection error
[image]

I also noticed that on my esp32 cam, esp32-S(i bought this on aliexpress) is written on the board itself, does this have something to do with errors, maybe something else should be put in the "tools->board" tab?
[image]

1)is there a problem with the wiring and power supply? - everything seems to be perfectly compiled and uploaded.
2)maybe the board is selected incorrectly?
3) I've missed something in the properties of the wifi I'm trying to join

[image]
[image]
[image]
[image]
[image]
[image]
[image]

s60sc, 2023-03-26:
Only works with sd card.
for streaming use this [app](https://github.com/easytarget/esp32-cam-webserver)

DJprasenjit, 2023-03-30:
can i use hostname insted of ip address to open the web server?
mdns name services in order to use http://[Host Name] instead of ip address.
please tell me this function works or not. i am having issue to find ip.

than-sad, 2023-03-30:
Hi, I'm using it from the beginning and I can confirm you is working perfect. Also you can setup a router rule to make the DHCP server assign always the same ip from the MAC adress.

Regards.

DJprasenjit, 2023-03-30:
> Hi, I'm using it from the beginning and I can confirm you is working perfect. Also you can setup a router rule to make the DHCP server assign always the same ip from the MAC adress.
> 
> Regards.

I tried it. set Host name "esp32cam" then go to browser type http://esp32cam or esp32cam.local but not working. what i did wrong? what else do i need to do? i also tried to set static ip but that system also not works, can you plesae explain? i am noob but learn fast

than-sad, 2023-03-30:
In my case works with no problem as you can see in the capture.
[image]

DJprasenjit, 2023-03-31:
> In my case works with no problem as you can see in the capture. [image]

is this your modified programme? please share me that. its looks different and cool. is this any app or only browser? i tried the same as capture you gave me. but not working bro, not in chrome on pc or mobile

than-sad, 2023-03-31:
Hi @DJprasenjit, the screen capture is the @s60sc project with just few cosmetic changes in MJPEG2SD.htm in safari for iPhone (I don't like Microsoft/google companies). It's still in WIP and I will share with @s60sc first when finished as the project owner.

I think your problem can be related to your network/router.

@atesis, the ESP32-CAM can work at 3,3 and 5 Volts. If you power it up with 5V in the 3,3 pin you will just see flash light on and not working, don't worry to burn it. Using TTL you have to specify the VCC (output voltage) with 3,3 or  5 V. you can use 3,3 and connect to 3,3 pin in the ESP32-CAM or 5 V. connecting to 5 V. pin. Your schema is correct but the HTML file loaded when you access to ESP32-CAM is located in the SD card, if no SD card, you can't run the web app.

Regards, Than.

s60sc, 2023-03-31:
An alternative to not using SD card is to store htm and config file on flash but don't try and record anything. In `appGlobals.h `change
`#define STORAGE SD_MMC`
To
`#define STORAGE LittleFS`

You will need to go thru installation process using AP connection.

s60sc, 2023-03-31:
You do have a connection - you received time from NTP server.
Your problem is no SD card installed.

s60sc, 2023-03-31:
I'm confused by what the problem is. If there is a problem with an underscore in the default host name, just change it.

s60sc, 2023-04-01:
you can change it yourself

[image]

closed 2023-04-16 by s60sc