s60sc/ESP32-CAM_MJPEG2SD #189: PSRAM Error

maximagination1, 2023-03-21:
"Startup Failure: Need PSRAM to be enabled"

Keeps throwing this error message in a blank page when opening the web server. I've already enabled it before uploading the code?

s60sc, 2023-03-21:
The esp hardware is reporting that psram is either not present or not enabled.

In arduino: Tools -> PSRAM "Enabled"

maximagination1, 2023-03-21:
Thanks for your reply.

I have already had PSRAM enabled in Tools before uploading the code, plus
the camera was just working fine for a good length of time before this
happened. I don't understand why.

Could this mean that my ESP32-CAM board's PSRAM chip has gotten fried or
stopped working?

I've tried over and over several times with other ESP32-CAM boards and I'm
getting the same issue.

Look forward to your reply.
Thanks.

s60sc, 2023-03-21:
Check you have a good power supply

Try uploading this sketch:

```
void setup() {
  printf("Total PSRAM: %d\n", ESP.getPsramSize());
}
void loop() {}
```

maximagination1, 2023-03-21:
Even though I just now managed to get it working through another method, thanks for your help anyway!

Clearing the MicroSD config file and starting over with copy-pasting a new one to the card did the trick in getting my camera's web server back up and running.

This PSRAM error came about after saving pin configurations including GPIOs that didn't seem to work with a microphone I was setting up. Would you please recommend the best 3 available ESP32-CAM pins to use for an I2S Mic? Thx.

Cheers,
Max.

s60sc, 2023-03-21:
Did you assign pin 16 - not usable as reserved for psram.

[Configure Web Page](https://github.com/s60sc/ESP32-CAM_MJPEG2SD#configuration-web-page) section indicates available pins, but as per [Audio Recording](https://github.com/s60sc/ESP32-CAM_MJPEG2SD#audio-recording) section its not worthwhile using a mic unless you have an S3

s60sc, 2023-03-21:
yes, but see [Configure Web Page](https://github.com/s60sc/ESP32-CAM_MJPEG2SD#configuration-web-page)  section for caveats on 12 and 33.  
Or upgrade to one of these [ESP32-S3 cam](https://www.aliexpress.com/item/1005004960637276.html?spm=a2g0o.productlist.main.1.8e8111600265pm&algo_pvid=52f309f2-2114-4ba1-aefe-5fbac0535d9c&algo_exp_id=52f309f2-2114-4ba1-aefe-5fbac0535d9c-0&pdp_ext_f=%7B%22sku_id%22%3A%2212000031163496708%22%7D&pdp_npi=3%40dis%21GBP%219.41%219.41%21%21%21%21%21%402122457116794222665611860d0713%2112000031163496708%21sea%21UK%211887864838&curPageLogUid=KASQoMvcpsvT)

closed 2023-04-16 by s60sc