s60sc/ESP32-CAM_MJPEG2SD #191: Clients

Makss8, 2023-03-25:
is it possible to have a preview of the camera image on 2 devices at the same time?  (I'm not talking about using clouds or servers, is there an option to change it in the code?)

s60sc, 2023-03-26:
not understood

Makss8, 2023-03-26:
I use the TinyCam monitor application to view the esp32 cam.  When I have live view from the camera on one device, the image cannot be loaded on the other device.  I have to exit from one device and then the preview will be displayed on the other device.  The camera does not allow you to view the image on several devices at the same time

s60sc, 2023-03-26:
you could change the following line in `appGlobals.h` but its asking a lot of a little microcontroller so will slow down and may crash

`#define MAX_CLIENTS 2 // allowing too many concurrent web clients can cause errors`

Makss8, 2023-03-26:
That's what I meant, I love your code, thank you!

Makss8, 2023-03-26:
Something's not working for me. I turned on 2 image broadcasts at once and only 1 is displayed. (in the code I set #define MAX_CLIENTS 3 // allowing too many concurrent web clients can cause errors) 
[image]
[image]
[image]

Makss8, 2023-03-26:
.

s60sc, 2023-03-26:
Reviewing the code its not designed for multiple streams as too complex

Makss8, 2023-03-26:
So I can't have a DVR connected to the camera and look at the camera view on the phone at the same time, sad.