s60sc/ESP32-CAM_MJPEG2SD #192: Pins Usage

maximagination1, 2023-03-30:
Despite knowing the mentioned IO pins that are available on the ESP32-CAM for external sensors/loads, I'd still like to know if there are 3 pins apart from IO12, IO4 and IO33 that can be used to run a microphone. 

Would pins IO3 (RX), IO1 (TX) and IO0 (- Strapping pin) work as regular IO pins for an I2S Mic? If so, could you please advise how it is best wired with the 3 pins? If it's not possible, do you know how else I can get an I2S Mic connected to the ESP32-CAM AI thinker board without an IO extender chip? Considering there's already a couple of servo motors and a PIR motion sensor taking up 3 pins.

Your advice would be appreciated. Thanks.

maximagination1, 2023-03-30:
> io0 is used by cam. io1 is outputting the logging. io3 can be used for
> input only - I use it for a ds18b20. The extra cost of an Esp32-s3 cam with
> more io is not much more than having to buy an extender.

Thanks for your reply.

Ok, this means I can only attach something like a voltage divider wire for
battery voltage monitoring or a temp sensor.

I couldn't figure out how to program my ESP32-C3-WROOM-02-N4 for the IO
extender as it's pinout is different than of other ESP32 chips. It programs
with different Strapping pins, I ended up failing to get it to work due to
a lack of guidance on it.

s60sc, 2023-03-30:
voltage divider can only be used on pin 33 (only free adc)

maximagination1, 2023-03-31:
Got it. So can I instead have a PIR motion sensor hooked up to pin 3?

s60sc, 2023-03-31:
Should work, but disconnect pir during upload.
Your above comments dont mention pin 13 which is also available

maximagination1, 2023-03-31:
Ok, thanks. I'll keep that in mind.

Pin 13 didn't work as it seems to be used up by the microSD card during
use. The web server would not load and it gave an SD error.

closed 2023-04-16 by s60sc