s60sc/ESP32-CAM_MJPEG2SD #195: Suggestion

Makss8, 2023-03-30:
hi can you add an option to show the current time in any corner?  (in recordings and in live view)
[image]

s60sc, 2023-03-31:
technically possible, but would require image processing by esp32 which would massively slow down frame rate. 
jpeg -> rgb -> add time -> jpeg

s60sc, 2023-03-31:
,with regrard to this issue and your issue #191, you could use a PC as a proxy to receive the cam stream, apply a timestamp,  display the stream, then forward to the dvr. You would have to research it yourself though.

s60sc, 2023-03-31:
see also this [fork](https://github.com/gemi254/ESP32-CAM_MJPEG2SD)

s60sc, 2023-04-14:
I can add feature to deep sleep at night and wake on LDR

s60sc, 2023-04-16:
sleep / wake in version 8.6.3

closed 2023-04-16 by s60sc