s60sc/ESP32-CAM_MJPEG2SD #196: Insert SVG icons from SD card in MJPPEG2SD.htm

than-sad, 2023-04-01:
Hi,

I'm working in the IU redesign and I want to use SVG icons. I copied it to /data/icons/*.svg and called it in the htm like [image] but the /data folder contents aren't visible getting 404.

@s60sc how can I do it?

Regards.

s60sc, 2023-04-01:
the request for content in the /data folder is handled by the web server, so for example to retrieve `configs.txt` write `src='/web?configs.txt'`
Dont think it will handle a folder hierarchy but you could try
See `webHandler(httpd_req_t* req)` in` webserver.cpp`

than-sad, 2023-04-01:
Yes, again you are right. Copied to / and updated calls in htm and works. I wanted in /icons folder to don't mees too much. I will check the weberserver.cpp in the future to make it handle /

Thanks again ;)

closed 2023-04-01 by than-sad