s60sc/ESP32-CAM_MJPEG2SD #197: Space removed from wifi ssid

gramtech, 2023-04-01:
Hi 

There are essentially two issues:
1. in the code, utils.cpp yo can set the wifi creds but they just seem to be ignored ???
2. (This is the one I am actually writing this for) - I am finding a space in the SSID name is being removed.

So for example if I go into the config after the AP is online (192.168.4.1) and then setting and set my wifi SSID and pass. after saving and rebooting I can see the space was remove.

Lets say I see the SSID to "My HomeWifi"

The following appears in the serial monitor view

[00:00:01.324 onWiFiEvent] Wifi Station started, connecting to: MyHomeWifi

any ideas on this?

than-sad, 2023-04-01:
Hi @gramtech, are you following the Readme instructions? In my case, after compile, started the ESP32-CAM, connect to AP, opened the browser pointing to 192.168.4.1 has instructions mention, insert my Wifi name and password, restart and done.

Why do you want to edit utils.cpp?
Regards.

s60sc, 2023-04-01:
[image]

than-sad, 2023-04-01:
Wooo jajajajaj @s60sc i didn't know what that function was. You think in all possibilities, great job!

s60sc, 2023-04-01:
Due to a previous request but I dont recommend it

than-sad, 2023-04-01:
Yes, I read everywhere using spaces in SSID name can cause problems. That's why I use Home, Home_5G and Home_6G

gramtech, 2023-04-01:
Hi, thanks for your reply.
I did see and try the use white space option but didnt think it worked. Could try again.

edwardfuzy, 2023-04-03:
Thanks for the above thread. I'm having the same issue.
Everytime the cam switches off and powers back on - the SSID resets with the correct name, but without spaces. I have tried to allow whitespaces but alas, no luck. The setting change is saved, but the SSID still reboots without spaces. Once I log into the AP and manually change it, then it works.

As I'm looking to load this onto nearly 10 ESP32 Cams - most of which will be perched out of reach, I need it to be reliable. Additionely, I'm wanting to integrate them into my HA - so I'm hoping a solution finds its way to me...

FYI - I'm a complete noob, have zero coding experience and am learning as I go along, so forgive my ignorance. That said, I can follow instructions, so any advice would be immensely appreciated.

Thanks for all the hard work, hopefully, there is a solution without changing my router name - I have over 135 esp's, devices and smarthome IOT's that would need changing...

s60sc, 2023-04-03:
I'll recheck the code

s60sc, 2023-04-04:
there is an issue with the order that the config file is loaded.
As an interim workaround, in file `prefs.cpp` change:
`bool allowSpaces = false;`
to:
`bool allowSpaces = true;`

gramtech, 2023-04-04:
@s60sc 

Thanks, I have tested and it is working. 

Appreciate your fast responses. 

G

closed 2023-04-16 by s60sc