s60sc/ESP32-CAM_MJPEG2SD #198: FTP auto-upload function is not valid on TimeLapse.

lztz0022, 2023-04-02:
Sir s60sc,
In version 8.61, I have found a situation that I didn't know if it was a bug: When I use Time-Lapse function, the FTP automatic upload function seems to fail. Check the log and there are no records of any attempts to connect FTP. I try to read codes but obviously, I haven't found any solution. (-_-)so sad.

Another situation, I use the default settings and I checked the PIR auto lamp function, I found that when I turned on the room light, the LED that lit up in the dark automatically turned off, but when I turned off the room light and returned to the dark, the LED did not auto light up again. I think you may have started optimizing PIR auto lamp after Issue #183, so I report this.

s60sc, 2023-04-02:
Auto upload isnt implemented for timelapse but I will look at adding it

Autolamp reworked to only operate with PIR. Did you want to lamp to stay on all night independent of PIR?

lztz0022, 2023-04-03:
Thanks for your reply and awesome work.
Regarding autolamp function, as you said, most of the time relying on PIR is more appropriate. When testing this feature, I only considered transitions between light and dark in a short period of time, without considering the load of LED lights in a long dark scene.

closed 2023-04-06 by lztz0022