s60sc/ESP32-CAM_MJPEG2SD #199: All directories were deleted when action mode on SD min free is set to 1

MisiekDP, 2023-04-02:
Hey,
I noticed today that all my recordings were deleted when the amount of free space reached the limit to free up some memory. As a result, the program should have deleted the last **one** oldest directory, but the program deleted **all** the directories and freed up all the memory (8GB of free memory out of 8GB of total memory).
In addition, the program also wanted to delete paths such as: /System Volume Information and //FOUND.000.
When deleting the /System Volume Information directory, the program completely jammed, tried to delete the directory for hours and prevented any recording.

Could you please take a look at the problem in question?

PS. I was also wondering if it is possible for FPT to connect as a client to ESP32-CAM to download multiple files/directories at once? I know that the ESP32-CAM allows you to upload files automatically, but there are so many files that it would be more convenient to download all the files to the computer and browse them then.

Logs from SD Card are in attachment.
[Logs ESP32-CAM.txt](https://github.com/s60sc/ESP32-CAM_MJPEG2SD/files/11133104/Logs.ESP32-CAM.txt)

s60sc, 2023-04-03:
Are you using lastest app version?

The esp32 could act as a ftp server, but the app is already complex and difficult to maintain so its not something that I want to add, although someone else could try.

s60sc, 2023-04-03:
There was a bug which should now be fixed, sorry

MisiekDP, 2023-04-03:
Yes i used the latest version ;)
Thanks!

closed 2023-04-03 by MisiekDP