s60sc/ESP32-CAM_MJPEG2SD #212: Start stream does not resume after a recording initiated by motion has ended

YapCK45, 2023-04-21:
V 8.6.3
1. **Start Stream** is initiated by clicking on the button and it changes to a blinking **Stop Stream** status. The display shows the current scene.
2. Motion detected starts capture and the **Start Recording** button begins to blink. When there is no more motion, recording ends and the status changes to a static **Start Recording**. The **Stop Stream** is still blinking but the display no longer shows the video. The status should change back to **Start Stream** as there is no longer any  motion.
3a. The **Start Recording** starts to blink even though there is no motion and this goes on indefinitely taking up storage space on the SD card. You can see the captured files on the SD card and there is no motion in them.
3b. If the **Start Recording** is clicked, it changes to **Stop Recording** and if it is clicked again, Recording apparently stops. But it starts again on its own without any motion and as in 3a, capturing files on the SD card without motion.
4. Eventually, this cycle stops or if the system is reset.

This observation can be replicated by having the web page on one half of the screen and the serial monitor on the other half of the screen as shown here:
[image]

The SD card is fast filling up as shown here:
[image]

s60sc, 2023-04-21:
The stream freezes as the esp32 is busy recording, but only stops when requested by user.
I wasnt able to recreate the repeated motion issue.

YapCK45, 2023-04-21:
Perhaps you could check on the recorded video files. I noticed that many of them did not have motion in them and I was wondering why. I would expected **every one** to start with motion unless they are initiated manually.

YapCK45, 2023-04-23:
I want to elaborate on the earlier comment to look at the recorded files. Below is a collection of files captured on 20230423. I **playback the first three files** and noticed that they looked **exactly the same**. 
[image]

At first, I thought this confirms my earlier comment. Then I suspected something is not right as there could be another possibility. So I **downloaded them** and found that the playback is different. So there is indeed motion capture.
Seems like the playback is not always playing back the file selected which led to my wrong conclusion. Apologies for that.

I did a ESP reboot and the playback is working as expected. Seems like somehow there is a issue in the file selection for playback. This is easily replicated but takes time to download the files for confirmation:

1. Select a few sequential files that are recorded for playback. If they show different scenes, then everything is working properly.
2. If they look the same, download them and playback to verify if there is motion and the scenes are different. If they are, it means the playback has some issue.

s60sc, 2023-04-23:
Cant recreate playback issue

closed 2023-05-04 by s60sc