s60sc/ESP32-CAM_MJPEG2SD #213: Cannot connect to server .......

pratik001010, 2023-04-25:
After uploading code to the board , we cannot connect to the server as long as tthe sd card is installed on the board , if sd card is ejected from the board then only ip is loaded with error message saying check your sd card

s60sc, 2023-04-25:
is the data folder downloaded to sd card?

pratik001010, 2023-04-26:
Yes sir ,, still cannot figure out why my board cannot connect to server without ejecting sd card

lefake, 2023-05-01:
Getting the same issue... 

This is with the SD Card (with the data folder)
[image]

This is without it
[image]

closed 2023-05-04 by s60sc