s60sc/ESP32-CAM_MJPEG2SD #217: new release - Arduino IDE 2.1.0 cc1plus: some warnings being treated as errors

krschneider, 2023-04-28:
FQBN: esp32:esp32:esp32cam
Using board 'esp32cam' from platform in folder: /Users/kevin_schneider/Library/Arduino15/packages/esp32/hardware/esp32/2.0.8
Using core 'esp32' from platform in folder: /Users/kevin_schneider/Library/Arduino15/packages/esp32/hardware/esp32/2.0.8
         .
         .
         .
/code/ESP32-CAM_MJPEG2SD/mqtt.cpp: In function 'void mqttPublish(const char*)':
/code/ESP32-CAM_MJPEG2SD/mqtt.cpp:35:60: error: '/status' directive writing 7 bytes into a region of size between 1 and 64 [-Werror=format-overflow=]
   if (!strlen(mqttPublishTopic)) sprintf(mqttPublishTopic, "%s%s/status", mqtt_topic_prefix, hostName);
                                                            ^~~~~~~~~~~~~
/code/ESP32-CAM_MJPEG2SD/mqtt.cpp:35:41: note: 'sprintf' output 8 or more bytes (assuming 71) into a destination of size 64
   if (!strlen(mqttPublishTopic)) sprintf(mqttPublishTopic, "%s%s/status", mqtt_topic_prefix, hostName);
                                  ~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/code/ESP32-CAM_MJPEG2SD/mqtt.cpp: In function 'void startMqttClient()':
/code/ESP32-CAM_MJPEG2SD/mqtt.cpp:181:22: error: '/lwt' directive writing 4 bytes into a region of size between 1 and 64 [-Werror=format-overflow=]
   sprintf(lwt_topic, "%s%s/lwt", mqtt_topic_prefix, hostName);
                      ^~~~~~~~~~
/code/ESP32-CAM_MJPEG2SD/mqtt.cpp:181:10: note: 'sprintf' output 5 or more bytes (assuming 68) into a destination of size 64
   sprintf(lwt_topic, "%s%s/lwt", mqtt_topic_prefix, hostName);
   ~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/Users/kevin_schneider/Documents/Arduino/krs code/ESP32-CAM_MJPEG2SD/mqtt.cpp:182:22: error: '/cmd' directive writing 4 bytes into a region of size between 1 and 64 [-Werror=format-overflow=]
   sprintf(cmd_topic, "%s%s/cmd", mqtt_topic_prefix, hostName);
                      ^~~~~~~~~~
/code/ESP32-CAM_MJPEG2SD/mqtt.cpp:182:10: note: 'sprintf' output 5 or more bytes (assuming 68) into a destination of size 64
   sprintf(cmd_topic, "%s%s/cmd", mqtt_topic_prefix, hostName);
   ~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1plus: some warnings being treated as errors

s60sc, 2023-04-28:
Version 8.6.5 should fix

krschneider, 2023-04-29:
thanks, s60cc - I got a clean compile with version 8.6.5 I really like this application, and commend you for your hard work to maintain it.

closed 2023-04-29 by krschneider