s60sc/ESP32-CAM_MJPEG2SD #219: XIAO ESP32S3 sense sketcch usage

FrancescoMontedori, 2023-05-02:
Hi, I recently purchased the new XIAO ESP32S3 sense board from SeedStudio
[image]
I would like to use your sketch with this board, as it has the ESP32S3 chip (should be good enough to obtain good video quality) and it has a internal I2C microphone.

I tried to make a slight modification to your sketch:
- I added the pin definition of the new board inside the file "camera_pins.h".
[image]

- I defined the CAM_BOARD in the "ESP32-CAM_MJPEG2SD":
[image]

- I have defined the CAM_BOARD in the main:However, once the sketch is loaded, when I open the serial monitor I get the following output:

[image]

This as the error suggests should be due to an incorrect pin definition for the camera and SD card.

I also attach the pin diagram of the board and the link to the datasheet:
[image]
[image]

Link for datasheet and useful information
https://wiki.seeedstudio.com/xiao_esp32s3_getting_started

Any suggestions on how to make your sketch work with this new board?

s60sc, 2023-05-02:
in `appGlobals.h `is the following:

```
#if defined(CAMERA_MODEL_ESP32S3_EYE)
// pins configured for SD card on this camera board
#define SD_MMC_CLK 39 
#define SD_MMC_CMD 38
#define SD_MMC_D0 40
#endif
```

You will need to add a separate section for XIAO ESP32S3, identifying which pins are being used for the SD card.

If you let me know what they are I can add this detail for anyone else who uses this board

FrancescoMontedori, 2023-05-02:
Unfortunately, it is a product that came out a few days ago and as a result the documentation is still a bit sparse. I was able to find this information on getting started.
They do not mention the CLK,CMD and D0 PINs in the datasheet.

[image]

Do you have any idea which PINs in the image I uploaded match the ones of your code?

[image]

s60sc, 2023-05-02:
Try
```
if defined(CAMERA_MODEL_XIAO_ESP32S3)
// pins configured for SD card on this camera board
#define SD_MMC_CLK 7 
#define SD_MMC_CMD 9
#define SD_MMC_D0 8
#endif
```

Also for cam error, did you comment out `#define CAMERA_MODEL_ESP32S3_EYE // Has PSRAM `

```
#elif defined(CONFIG_IDF_TARGET_ESP32S3)
// default pin configuration below for Freenove ESP32S3 cam boards
#define CAMERA_MODEL_ESP32S3_EYE // Has PSRAM
#endif
```

FrancescoMontedori, 2023-05-02:
Ok with the PIN configuration you suggested, the sketch works smoothly, the video recording starts correctly when motion is detected, and the file is saved with the correct FPS. 

Thanks a lot.

Now i want to try to use also the built in I2C microphone to record the audio and video at the same time (hopefully).
I will try a little bit with my own strength/resources to try to solve this problem. In case I get stuck I will probably contact you again!

s60sc, 2023-05-03:
Out of curiosity, I looked up the mic details and it is a PDM mic. The problem is that PDM is only supported by I2S0 which is already used by the espressif camera library. I2S1 is available but only supports I2S mics (as per my readme). I've left a comment on the Seeed forum.

s60sc, 2023-05-03:
Have since found out that although camera lib for ESP32 uses I2S0 DMA, for ESP32S3 it uses GDMA, so PDM mic can be used simultaneously. However the app would need updating to support PDM mic.

rws-githubuser1, 2023-07-09:
Hi, I am also using the XIAO ESP32S3 sense board but although I've set "#define CAMERA_MODEL_XIAO_ESP32S3" and have followed your instructions in https://github.com/s60sc/ESP32-CAM_MJPEG2SD/issues/219#issuecomment-1531898763, I'm still getting the following error messages:

13:17:51.429 -> [17:17:51.834 setup] =============== Starting ===============
13:17:51.429 -> E (1188) sdmmc_req: sdmmc_host_wait_for_event returned 0x107
13:17:51.429 -> E (1188) sdmmc_sd: sdmmc_check_scr: send_scr returned 0x107
13:17:51.429 -> E (1189) vfs_fat_sdmmc: sdmmc_card_init failed (0x107).
13:17:51.429 -> [17:17:53.006 ERROR @ utilsFS.cpp:69] SD card mount failed 
13:17:51.630 -> [17:17:53.183 prepCam] Camera init OK for model OV2640 on board CAMERA_MODEL_XIAO_ESP32S3
13:17:51.664 -> [17:17:53.230 loadConfig] Load config
13:17:51.664 -> [17:17:53.230 ERROR @ prefs.cpp:133] Failed to load file /data/configs.txt 

Any ideas?

s60sc, 2023-07-09:
in `appGlobals.h`

/**************************************************************************
 Uncomment one only of the camera models below if not using a default above
 **and comment out above define block**
 Selecting wrong model may crash your device due to pin conflict
***************************************************************************/

rws-githubuser1, 2023-07-09:
Thanks for the quick response. The following are the first few lines in appGlobals.h which appears to have resolved the issue. Thanks for the great support!

//#if defined(CONFIG_IDF_TARGET_ESP32)
//// default pin configuration for ESP32 cam boards
//#define CAMERA_MODEL_AI_THINKER // Has PSRAM  
//#elif defined(CONFIG_IDF_TARGET_ESP32S3)
//// default pin configuration below for Freenove ESP32S3 cam boards
//#define CAMERA_MODEL_ESP32S3_EYE // Has PSRAM 
//#endif

/**************************************************************************
 Uncomment one only of the camera models below if not using a default above
 and comment out above define block
 Selecting wrong model may crash your device due to pin conflict
***************************************************************************/

//#define CAMERA_MODEL_WROVER_KIT // Has PSRAM
//#define CAMERA_MODEL_ESP_EYE // Has PSRAM
//#define CAMERA_MODEL_M5STACK_PSRAM // Has PSRAM
//#define CAMERA_MODEL_M5STACK_V2_PSRAM // M5Camera version B Has PSRAM
//#define CAMERA_MODEL_M5STACK_WIDE // Has PSRAM
//#define CAMERA_MODEL_M5STACK_ESP32CAM // No PSRAM
//#define CAMERA_MODEL_M5STACK_UNITCAM // No PSRAM
//#define CAMERA_MODEL_TTGO_T_JOURNAL // No PSRAM
#define CAMERA_MODEL_XIAO_ESP32S3 // Has PSRAM
//#define CAMERA_MODEL_ESP32_CAM_BOARD
//#define CAMERA_MODEL_ESP32S2_CAM_BOARD
//#define CAMERA_MODEL_ESP32S3_CAM_LCD

closed 2023-07-28 by s60sc

maxkcy, 2023-09-29:
I am experiencing the same issue "Startup Failure: Check SD card inserted". I have a Aideepen model (commonly available on Amazon, and deducting from the images on their website it is an AI-Thinker ESP32S). The example camera sketch from AI-Thinker board works on this model.

I made sure the top part was not commented out, and that no other cam is uncommented in appGlobals.h, nonetheless I'm still getting the "Startup Failure: Check SD card inserted".

maxkcy, 2023-10-01:
> I am experiencing the same issue "Startup Failure: Check SD card inserted". I have a Aideepen model (commonly available on Amazon, and deducting from the images on their website it is an AI-Thinker [...]

[Resolved] Turns out it was a SD card issue. I bought one of alibaba for less than a dollar, 8 GB SD card, and seems incompatible with the ESP32 even though it works fine for file storage. I am not sure what the reason for this is, is it the size? The issue was fixed when I decided to put in a different 32 GB card and it worked.

klimentiy23, 2023-11-25:
Stream CAMERA_MODEL_AI_THINKER doesn't work

SamMcCormick61, 2023-12-12:
I got mine to be recognized by setting the FAT32 allocation unit to 8192

[image]

Achllle, 2025-02-23:
Did not expect changing the allocation unit size (AUS) to work but it did! on a Linux PC:
`sudo mkfs.fat -s 16 /dev/sdaX` (8192/512 = 16)

Mech4Life1970, 2025-09-21:
> I got mine to be recognized by setting the FAT32 allocation unit to 8192
> 
> [image]

not sure if you even refer back to this anymore being last post from 2023 but your screenshot shows a NOT 4gb sd card did that really work just by changing the allocation size if so did it mess anything else up?? curious cuz i was just thinking that this cam will be left at days on end and i want it to record everything ya kno..