s60sc/ESP32-CAM_MJPEG2SD #221: Failed to ping gateway, restart wifi ...

DJprasenjit, 2023-05-05:
[08:34:27.875 WARN pingTimeout] Failed to ping gateway, restart wifi ... 
[08:34:27.875 setWifiSTA] Wifi Station IP from DHCP
[08:34:27.879 onWiFiEvent] WiFi Station disconnected
.[08:34:27.952 onWiFiEvent] WiFi Station connection to FREE_WIFI-ZONE, using hostname: ESP-CAM_MJPEG_04CAC415B80C
..[08:34:28.993 onWiFiEvent] Wifi Station IP, use 'http://192.168.0.106' to connect
[08:34:35.637 startWifi] Wifi stats for FREE_WIFI-ZONE - signal strength: -38 dBm; Encryption: WPA_WPA2_PSK; channel: 11
i used external antena and good power then also  Failed to ping gateway, restart wifi ...  coming continuosly
signal strength: -38 dBm

DJprasenjit, 2023-05-05:
I changed the wifi router and also same issue. why this? i have a good signal strength , then this coming continuosly

[08:48:08.681 WARN pingTimeout] Failed to ping gateway, restart wifi ... 
[08:48:08.682 setWifiSTA] Wifi Station IP from DHCP
[08:48:08.685 onWiFiEvent] WiFi Station disconnected
.[08:48:08.699 onWiFiEvent] Wifi Station lost IP
[08:48:08.740 streamHandler] MJPEG: 198 frames, total 3121kB in 15.7s @ 12.6fps
.[08:48:09.226 onWiFiEvent] WiFi Station connection to LuKey4177, using hostname: ESP-CAM_PK
[08:48:09.249 onWiFiEvent] Wifi Station IP, use 'http://192.168.137.213' to connect
[08:48:16.140 startWifi] Wifi stats for LuKey4177 - signal strength: -12 dBm; Encryption: WPA2_PSK; channel: 1
[08:48:38.679 WARN pingTimeout] Failed to ping gateway, restart wifi ... 
[08:48:38.680 setWifiSTA] Wifi Station IP from DHCP
[08:48:38.683 onWiFiEvent] WiFi Station disconnected
..[08:48:39.240 onWiFiEvent] WiFi Station connection to LuKey4177, using hostname: ESP-CAM_PK
[08:48:39.263 onWiFiEvent] Wifi Station IP, use 'http://192.168.137.213' to connect
[08:48:46.140 startWifi] Wifi stats for LuKey4177 - signal strength: -11 dBm; Encryption: WPA2_PSK; channel: 1
[08:49:08.679 WARN pingTimeout] Failed to ping gateway, restart wifi ... 
[08:49:08.680 setWifiSTA] Wifi Station IP from DHCP
[08:49:08.683 onWiFiEvent] WiFi Station disconnected
.[08:49:08.722 onWiFiEvent] WiFi Station connection to LuKey4177, using hostname: ESP-CAM_PK
[08:49:08.746 onWiFiEvent] Wifi Station IP, use 'http://192.168.137.213' to connect
[08:49:15.840 startWifi] Wifi stats for LuKey4177 - signal strength: -13 dBm; Encryption: WPA2_PSK; channel: 1
[08:49:38.679 WARN pingTimeout] Failed to ping gateway, restart wifi ... 
[08:49:38.680 setWifiSTA] Wifi Station IP from DHCP
[08:49:38.683 onWiFiEvent] WiFi Station disconnected
.[08:49:38.719 onWiFiEvent] WiFi Station connection to LuKey4177, using hostname: ESP-CAM_PK
[08:49:38.737 onWiFiEvent] Wifi Station IP, use 'http://192.168.137.213' to connect
[08:49:45.638 startWifi] Wifi stats for LuKey4177 - signal strength: -14 dBm; Encryption: WPA2_PSK; channel: 1
[08:50:08.679 WARN pingTimeout] Failed to ping gateway, restart wifi ... 
[08:50:08.680 setWifiSTA] Wifi Station IP from DHCP
[08:50:08.683 onWiFiEvent] WiFi Station disconnected
.[08:50:08.715 onWiFiEvent] WiFi Station connection to LuKey4177, using hostname: ESP-CAM_PK
[08:50:08.741 onWiFiEvent] Wifi Station IP, use 'http://192.168.137.213' to connect
[08:50:15.839 startWifi] Wifi stats for LuKey4177 - signal strength: -13 dBm; Encryption: WPA2_PSK; channel: 1
[08:50:38.679 WARN pingTimeout] Failed to ping gateway, restart wifi ... 
[08:50:38.680 setWifiSTA] Wifi Station IP from DHCP
[08:50:38.683 onWiFiEvent] WiFi Station disconnected
.[08:50:38.697 onWiFiEvent] Wifi Station lost IP
[08:50:38.733 onWiFiEvent] WiFi Station connection to LuKey4177, using hostname: ESP-CAM_PK
[08:50:38.764 onWiFiEvent] Wifi Station IP, use 'http://192.168.137.213' to connect
[08:50:45.638 startWifi] Wifi stats for LuKey4177 - signal strength: -13 dBm; Encryption: WPA2_PSK; channel: 1
[08:51:08.679 WARN pingTimeout] Failed to ping gateway, restart wifi ... 
[08:51:08.680 setWifiSTA] Wifi Station IP from DHCP
[08:51:08.683 onWiFiEvent] WiFi Station disconnected
..[08:51:09.225 onWiFiEvent] WiFi Station connection to LuKey4177, using hostname: ESP-CAM_PK
[08:51:09.281 onWiFiEvent] Wifi Station IP, use 'http://192.168.137.213' to connect
[08:51:16.138 startWifi] Wifi stats for LuKey4177 - signal strength: -11 dBm; Encryption: WPA2_PSK; channel: 1
[08:51:38.679 WARN pingTimeout] Failed to ping gateway, restart wifi ... 
[08:51:38.680 setWifiSTA] Wifi Station IP from DHCP
[08:51:38.683 onWiFiEvent] WiFi Station disconnected
..[08:51:39.217 onWiFiEvent] WiFi Station connection to LuKey4177, using hostname: ESP-CAM_PK
[08:51:39.237 onWiFiEvent] Wifi Station IP, use 'http://192.168.137.213' to connect
[08:51:46.138 startWifi] Wifi stats for LuKey4177 - signal strength: -12 dBm; Encryption: WPA2_PSK; channel: 1

s60sc, 2023-05-05:
Perhaps router not responding to pings. On windows command line enter `ping your_router_ip `and see if it responds

DJprasenjit, 2023-05-06:
Reply from 192.168.0.1: bytes=32 time=1ms TTL=64
router ping looks ok and when i connect it to my laptop 5g wifi hotspot same issue coming with my two cam module version 8.6.1 and 8.7 both version and board showing failed to ping.
but when i delete the wifi credincial and connect with AP mode AKA station mode there is no such issue coming. working very fine. but when ssid and pass giving that problem comes. hope you understand now

s60sc, 2023-05-06:
`[08:51:39.237 onWiFiEvent] Wifi Station IP, use 'http://192.168.137.213/' to connect `

is not the same gateway as 

`Reply from 192.168.0.1: bytes=32 time=1ms TTL=64`

Try `ping http://192.168.137.1`

AP is cam hotspot so doesnt need ping

DJprasenjit, 2023-05-07:
> [08:34:27.875 WARN pingTimeout] Failed to ping gateway, restart wifi ... [08:34:27.875 setWifiSTA] Wifi Station IP from DHCP [08:34:27.879 onWiFiEvent] WiFi Station disconnected .[08:34:27.952 [...]

this is the serial recieved when i connected with router. and the other ip showing that time i connected with my laptop's hotspot. and i pinged from my router

TheFrankmanDE, 2023-05-17:
Hi. 
I have the same issue. 3 AP's (Vodafone Router, Netgear Router and AP on mobile / smartphone), allways the same. Workaround (commenting startWifi). No access to the Cam. Cam not at accesspoint and no cam-AP available. Static IP has no effect too. 
Please help.

HAmamiya, 2023-05-18:
I did some "debugging" by modifying `static void pingTimeout(esp_ping_handle_t hdl, void *args)` in `utils.cpp` as follows:

```
static void pingTimeout(esp_ping_handle_t hdl, void *args) {
  LOG_WRN("Failed to ping gateway, restart wifi ...");
  //startWifi(false);
  IPAddress ipAddr = WiFi.gatewayIP();
  LOG_WRN("Restart wifi function is commented out by Hazuki. Gateway IP: %d.%d.%d.%d",ipAddr[0], ipAddr[1], ipAddr[2], ipAddr[3]);
}
```

I found that the gateway IP was correct. I *guess* the problem lies on routing under multiple interfaces/IPs (e.g. ping on AP(address) instead of station(address)). Since comment out `startWifi(false);` serves my purpose I guess I won't dig in the problem soon.

s60sc, 2023-05-18:
If the gateway is valid and ICMP enabled then mostly likely bad password

TheFrankmanDE, 2023-05-18:
Thanks for your reply.
ICMP is always enabled in all networks here at home. So pinging is not an issue here. The password is the right one on all different routers.
In the serial monitor I can see that the connection establishes at startup of the cam. After the cam trying to ping the AP the errors appear and the wifi will be resettet at the cam. If I coment out the "startWifi=false" the connection stays allive, but the cam does not response. There are no routingtables active for the IP of the cam. It does further not appear in the "Connected Devices"-list of the tried routers and even fixed IPs given to the cam doesn't change anything even i see in the serial monitor that the cam seems to be still connected. Some network scanner im using here doesn't even "see" the cam. But if I give another device the same IP I can see an IP-conflict with the same static IP.
Pinging the IP of the cam from other devices gives timeout. So I think that the problem is somewere in the cam. My program skills are not so good that I can find a solution. I'm a network admin, so I think I gave the right password while testing a lot before I wrote.
Regards

s60sc, 2023-05-18:
If the cam does not appear on the router it is because it has been unable to connect and therefore gets no ping response which is why it keeps retrying. The cam can report the ssid details but this does not mean a connection has been made.
> If I coment out the "startWifi=false" the connection stays allive, but the cam does not response.

The cam does not respond because there is no connection.

>Pinging the IP of the cam from other devices gives timeout.

The cam is not setup to repond to pings.

TheFrankmanDE, 2023-05-18:
Thanks for the important informations. I will try different things here and will inform you what's happening.
Thanks!

Magnum164, 2023-07-01:
Need to add my name to this issue. I can post log but it is the same as those above. I do know my connections are correct/router connection is fine as it works with other ESP32 camera code. Just for some reason I watch this on my router and after 30 seconds of connect it seems to disconnect station, the the IP is lost. 

It will reconnect, and simply keep retrying. 

Suggestions above do not work. However when playing with other webservers on these ESP32, the situation is similar to multiple web connections to the ESP32, as it can handle only one at a time as I understand? 

If I follow the code, it seems the ping starts at 30 seconds? If so could it be some chips are seeing the ping as multiple web connections? If so, what is quickest way to disable pinging to test. Haven't found an easy way yet.

TheFrankmanDE, 2023-07-01:
Hi.
Took other surveillance cam code from git. All worked. But at least I took the original ESP32-Cam firmware/software. Not that convenient like that here but I have no more time to waste for trying anymore...

closed 2023-07-09 by s60sc

FoxWhiskey, 2023-07-30:
As a matter of fact, I experience(d) the same issue, as well!

By coincidence I found, that the problem could be caused by the way, the system does DNS: while doing a `tcpdump`on the router's network interface I noticed an MDNS-packet sent to a multicast address in order to resolve the NTP-server hostname given in the config !
As my router does not support MDNS-queries, the request is never replied to, thus the hostname never resolved.
Replacing the NTP-server hostname by its IP-address then solved the issue!

Why name resolving the NTP-server hostname is related to the Wifi mechanism - I haven't researched.