s60sc/ESP32-CAM_MJPEG2SD #222: Timelaps has limited function

DJprasenjit, 2023-05-10:
There is only a button time laps on . But we need some extra features. Like manual start and stop time laps. We can set the interval.
But there must be a option within timelaps. Esp32 goes sleep and when the capture interval comes esp wake up and click photo and go to sleep. 
I am sure that if it goes sleep then Avi will not built. But its ok. You can only save the JpEG files. After we can build video from those pictures.
And please a option that can turn on and off flash light when time laps photo click.
Thanks. Please read all. And add these

s60sc, 2023-05-10:
Not something I want to implement, but someone else is welcome to try. 
I may consider the time lapse flash light.

DJprasenjit, 2023-05-15:
because we cant see there time laps running or interruped. flashing the Led will be help full then we can feel that time laps is on, there must be a manual stop button to end the time laps and save it

DJprasenjit, 2023-05-15:
Please add a timmer remaining time laps near the manual stop and save button. I am using it as 3d printer time laps. So can it possible to set a option time laps from selected time. Or from a switch clicks every time takes a photo. And when click stop and save then it will stop. Flashing the LED is Must

s60sc, 2023-07-09:
try latest version

closed 2023-07-28 by s60sc