s60sc/ESP32-CAM_MJPEG2SD #225: Feedback on version 8.7.1a, and a question about the ESP32cam MB

lztz0022, 2023-06-02:
Sir s60sc,
I've been downloading and using version 8.7.1a for 2 weeks, and it's the most stable version after I followed this project.
There was **no crashes** within 2 weeks. And cutting off the power during the time-lapse video will **not cause the SD card mount failed** when next restarting. The situation of **the continuous failure of pinging the gateway** leads to the waste of system resources **to restart WIFI has basically disappeared**. I think this version is perfect for the functions I use most.

A small question that may have nothing to do with this code, hope to get your advice: I plug the esp32cam into the data downloading mainboard for use. The power light on the mainboard will always be on, is it possible to turn off the power light on the mainboard through the program code? Thanks.

s60sc, 2023-06-02:
Thanks for feedback.
The power led is hardware controlled so cannot be changed by code.

lztz0022, 2023-06-02:
Got it. Thanks for reply.

closed 2023-06-04 by lztz0022