s60sc/ESP32-CAM_MJPEG2SD #227: How do I put the scheduling function in?

beodeulsoft, 2023-06-14:
In order to use this device on the desk in the office, I want to disable motion detection during business hours, so I want to put in a schedule function.
[image]
#MJPEG2SD.htm
<nav class="quick-nav" id="scheduler" name="weekly-schedule">?</nav>
...
<nav class="menu panel" id="weekly-schedule">
                <input type="checkbox" id="schedule-cb" class="menu-action">
                <div class="pin-menu" title="Pin / unpin menu">?</div>
                <label for="schedule-cb" class="nav-toggle">?&nbsp;&nbsp;Weekly Schedule&nbsp;&nbsp;</label>
                <div class='addButtons'></div>
                <div>
                  <div class="input-group" id="week0-group">
                    <label for="week0">Monday</label>
                    <div class="switch">
                      <input id="week0" type="checkbox" checked="checked">
                      <label title="Enable detecting day of the week" class="slider" for="week0"></label>
                    </div>
                  </div>
                  <div class="input-group" id="week1-group">
                    <label for="week1">Tuesday</label>
                    <div class="switch">
                      <input id="week1" type="checkbox" checked="checked">
                      <label title="Enable detecting day of the week" class="slider" for="week1"></label>
                    </div>
                  </div>
                  <div class="input-group" id="week2-group">
                    <label for="week2">Wednesday</label>
                    <div class="switch">
                      <input id="week2" type="checkbox" checked="checked">
                      <label title="Enable detecting day of the week" class="slider" for="week2"></label>
                    </div>
                  </div>
                  <div class="input-group" id="week3-group">
                    <label for="week3">Thursday</label>
                    <div class="switch">
                      <input id="week3" type="checkbox" checked="checked">
                      <label title="Enable detecting day of the week" class="slider" for="week3"></label>
                    </div>
                  </div>
                  <div class="input-group" id="week4-group">
                    <label for="week4">Friday</label>
                    <div class="switch">
                      <input id="week4" type="checkbox" checked="checked">
                      <label title="Enable detecting day of the week" class="slider" for="week4"></label>
                    </div>
                  </div>
                  <div class="input-group" id="week5-group">
                    <label for="week5">Saturday</label>
                    <div class="switch">
                      <input id="week5" type="checkbox" checked="checked">
                      <label title="Enable detecting day of the week" class="slider" for="week5"></label>
                    </div>
                  </div>
                  <div class="input-group" id="week6-group">
                    <label for="week6">Sunday</label>
                    <div class="switch">
                      <input id="week6" type="checkbox" checked="checked">
                      <label title="Enable detecting day of the week" class="slider" for="week6"></label>
                    </div>
                  </div>
                  <div class="vsep"></div>
                  <div class="input-group" id="day00-group">
                    <label for="day00">00:00 ~ 01:00</label>
                    <div class="switch">
                      <input id="day00" type="checkbox" checked="checked">
                      <label title="Enable detecting time of the day" class="slider" for="day00"></label>
                    </div>
                  </div>
                  <div class="input-group" id="day01-group">
                    <label for="day01">01:00 ~ 02:00</label>
[... 227 lines cut ...]
  vTaskNotifyGiveFromISR(captureHandle, &xHigherPriorityTaskWoken); // wake capture task to process frame
  if (isPlaying) xSemaphoreGiveFromISR (playbackSemaphore, &xHigherPriorityTaskWoken ); // notify playback to send frame
  if (xHigherPriorityTaskWoken == pdTRUE) portYIELD_FROM_ISR();
}
...
static void startSDtasks() {
  //init schedule
  for(int i=0; i < 7; i++)
      week[i] = true;
  for(int i=0; i < 24; i++)
      day[i] = true;
  // tasks to manage SD card operation
  xTaskCreate(&captureTask, "captureTask", 1024 * 4, NULL, 5, &captureHandle);
  xTaskCreate(&playbackTask, "playbackTask", 1024 * 4, NULL, 4, &playbackHandle);
  sensor_t * s = esp_camera_sensor_get();
  fsizePtr = s->status.framesize; 
  setFPS(frameData[fsizePtr].defaultFPS); // initial frames per second  
  debugMemory("startSDtasks");
}

After working on the above, I uploaded it to the esp32-s3 cam board, and the web server is not working properly. Even if I change the location of the getWeekTime function, I can't save the schedule information even if I press the save button.

#define MAX_CONFIGS 130 // > number of entries in configs.txt

I changed the above value but it still doesn't work, is there any other good way?

beodeulsoft, 2023-06-16:
It's hard because I'm not a web developer.
Thank you.

s60sc, 2023-06-20:
one thought, the `configs.txt` file will not be updated unless the data items already exist in it, so you need to ensure you have added the new data items in `configs.txt` in the correct format, eg:
`week0~0~98~~na `
where `week0` is var name, `0` is initial value (false)

beodeulsoft, 2023-06-21:
Thank you for your advise.

beodeulsoft, 2023-07-04:
Maybe it was a pointer problem before, but the mcu kept rebooting, but it works when I fix the function that checks the schedule as below.

int arrWeek[] = {1,1,1,1,1,1,1};
int arrDay[] = {1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1};

void* getWeekHour() {
   struct timeval tv;
   gettimeofday(&tv, NULL);

   return localtime(&tv.tv_sec);
}

static boolean processFrame() {
  if(timeSynchronized) //init ntp server?
  {
    // check weekly schedule 
    struct tm* timeinfo = (tm*)getWeekHour();
    int nWeek = timeinfo->tm_wday;
    if(arrWeek[nWeek] == 0) 
    {
      LOG_ERR("processFrame: Week %d is not scheduled", nWeek);
      //delay(100);
      return false;
    }
    int nHour = timeinfo->tm_hour;
    if(arrDay[nHour] == 0) 
    {
      LOG_ERR("processFrame: Time %d is not scheduled", nHour);
      //delay(100);
      return false;
    }
  } 
...
}

closed 2023-07-09 by s60sc