s60sc/ESP32-CAM_MJPEG2SD #229: Wifi connection issue

vanyayudin26, 2023-06-14:
[00:49:44 Web] Connect to: ws://192.168.4.1:80/ws
[00:49:44 Web] Connected
[18:49:45.423 wsHandler] Websocket connection: 56
[18:49:46.131 WARN pingTimeout] Failed to ping gateway, restart wifi ...
[18:49:46.134 setWifiSTA] Wifi Station IP from DHCP
[18:49:49.763 onWiFiEvent] WiFi Station disconnected
[18:49:51.142 WARN startWifi] SSID MI-R3GV2_SPI_537F unable to connect
[18:49:54.566 startWifi] Wifi stats for MI-R3GV2_SPI_537F - signal strength: -71 dBm; Encryption: WPA2_PSK; channel: 11
[18:50:16.131 WARN pingTimeout] Failed to ping gateway, restart wifi ...
[18:50:16.134 setWifiSTA] Wifi Station IP from DHCP
[18:50:19.762 onWiFiEvent] WiFi Station disconnected
[18:50:21.142 WARN startWifi] SSID MI-R3GV2_SPI_537F unable to connect
[18:50:24.966 startWifi] 
[image]
[image]
Wifi stats for MI-R3GV2_SPI_537F - signal strength: -72 dBm; Encryption: WPA2_PSK; channel: 11

chotakoebipki, 2023-06-14:
I can connect to other network (Android phone AP), but not for my home AP

s60sc, 2023-06-14:
wrong password?

chotakoebipki, 2023-06-15:
No, password is right, i also try set password in sketch.

closed 2023-07-09 by s60sc

SlavTiger, 2023-07-10:
i have the same problem. it asks for the router ip address in config, but which address? the ipv4? the ipv6 both temporary or link-local? it does not specify.

s60sc, 2023-07-10:
ipv4