s60sc/ESP32-CAM_MJPEG2SD #232: SD CARD Error

WalidQ, 2023-06-22:
I am using 4GB SD Card in ESP32 CAM. i uploaded code with changes in wifi credentials only. 

somehow i get these errors for ESP32 CAM 
E (63967) sdmmc_req: sdmmc_host_wait_for_event returned 0x107
E (63967) diskio_sdmmc: Check status failed (0x107)
E (64968) sdmmc_req: sdmmc_host_wait_for_event returned 0x107
E (64969) diskio_sdmmc: Check status failed (0x107)

on Web page it shows Startup Failure : Check SD Card inserted. 

SD card is perfectly working for other program and formatted using SD card Formatter.  on youtube video 4GB SD card was suggested so i am using 4GB SD card as well. 

[image]
[image]

what seems to be the issue here. do i need to modify app_secific.cpp ???

Also is this program able to store Audio and video both in Avi format?

s60sc, 2023-06-22:
hardware problem, may be pin conflict with your mic. Dont change code, change settings on web page edit config tab

audio and video stored in same avi file

codedogz777, 2023-06-28:
> hardware problem, may be pin conflict with your mic. Dont change code, change settings on web page edit config tab
> 
> audio and video stored in same avi file

how? that error would mean that he cant open the webpage no?
it would tell to check the sd card

WalidQ, 2023-07-03:
> hardware problem, may be pin conflict with your mic. Dont change code, change settings on web page edit config tab
> 
> audio and video stored in same avi file

I got two more ESP32 CAM boards and it turned out there was SD card Slot issue. Now it starts streaming video but i am not not getting any audio. i assign microphone pin numbers through web browser but as soon as controller restarts. configs file causes the error. 

Can i stream avi format directly to web browser? Audio and video both streamed to web browser?

WalidQ, 2023-07-03:
> hardware problem, may be pin conflict with your mic. Dont change code, change settings on web page edit config tab
> 
> audio and video stored in same avi file

I got two more ESP32 CAM boards and it turned out there was SD card Slot issue. Now it starts streaming video but i am not not getting any audio. i assign microphone pin numbers through web browser but as soon as controller restarts. configs file causes the error. this is pin number i assign in the web browser 

[image]

Can i stream avi format directly to web browser? Audio and video both streamed to web browser?

WalidQ, 2023-07-07:
[image]

to avoid this issue i changed board to ESP TTGO CAMERA PLUS but somehow it's again showing SD card initialize error. i noticed that sd card error remains with most of the boards. SD card are formatted using SD card formatter in FAT32. they are working fine. but failed to initialize. should i switch to older versions of sd card libraries?

s60sc, 2023-07-07:
This model uses a non standard pinout that is not currently implemented, I will aim to add it in the next 24 hours

s60sc, 2023-07-08:
New version uploaded
In `appGlobals.h`, comment out:

```
#if defined(CONFIG_IDF_TARGET_ESP32)
// default pin configuration for ESP32 cam boards
#define CAMERA_MODEL_AI_THINKER // Has PSRAM  
#elif defined(CONFIG_IDF_TARGET_ESP32S3)
// default pin configuration below for Freenove ESP32S3 cam boards
#define CAMERA_MODEL_ESP32S3_EYE // Has PSRAM
#endif

```
and uncomment:

`//#define CAMERA_MODEL_TTGO_T_CAMERA_PLUS`

As I dont have this board, you will have to confirm that it works

closed 2023-07-28 by s60sc