s60sc/ESP32-CAM_MJPEG2SD #234: SD Card  detected by esp32 cam but not with this specific code

codedogz777, 2023-06-28:
I always get these code when i try to run the firmware
E (509) sdmmc_common: sdmmc_init_ocr: send_op_cond (1) returned 0x107
E (510) vfs_fat_sdmmc: sdmmc_card_init failed (0x107).
[00:00:01.853 ERROR @ utilsFS.cpp:69] SD card mount failed 

ive checked using other codes that uses the sd card aswell and those worked fine, does anyone know how i can fix this?

s60sc, 2023-06-28:
`E (510) vfs_fat_sdmmc: sdmmc_card_init failed (0x107).` is a hardware error, nothing to do with app. 
Pin conflict or wrong camera model selected.

codedogz777, 2023-06-28:
> `E (510) vfs_fat_sdmmc: sdmmc_card_init failed (0x107).` is a hardware error, nothing to do with app. Pin conflict or wrong camera model selected.

wdym pin conflict
other codes work except for this one and i dont know why

codedogz777, 2023-06-28:
i also didnt change the camera model because i thought it was the default AI-Thinker ov2640 camera model

hubdogz777, 2023-06-29:
> `E (510) vfs_fat_sdmmc: sdmmc_card_init failed (0x107).` is a hardware error, nothing to do with app. Pin conflict or wrong camera model selected.

can u help please?

codedogz777, 2023-06-29:
> I always get these code when i try to run the firmware E (509) sdmmc_common: sdmmc_init_ocr: send_op_cond (1) returned 0x107 E (510) vfs_fat_sdmmc: sdmmc_card_init failed (0x107). [00:00:01.853 ERROR @ utilsFS.cpp:69] SD card mount failed
> 
> also
> 
> [00:00:11.359 ERROR @ ESP32-CAM_MJPEG2SD.ino:147] Startup Failure: Check SD card inserted
> 
> ive checked using other codes that uses the sd card aswell and those worked fine, does anyone know how i can fix this?
> 
> im using an esp32 cam mb to program it and using an 8 gb card with fat32 format that arleady has fata folder in it

im not sure if it even counts as a hardware problem, as other codes besides this one works, if i may is there any other code that can record while streaming and detect motion without a pir sensor aswell?

codedogz777, 2023-07-03:
> `E (510) vfs_fat_sdmmc: sdmmc_card_init failed (0x107).` is a hardware error, nothing to do with app. Pin conflict or wrong camera model selected.

ive checked using the SD_MMC code and it works just fine so i dont understand what u mean by how its because of hardware issue

#include "SD_MMC.h"
 
void setup(){
   
    Serial.begin(115200);
     
    if(!SD_MMC.begin()){
        Serial.println("Failed to mount card");
        return;
    }
    uint8_t cardType = SD_MMC.cardType();
 
    Serial.print("\nCard Type: ");
     
    if(cardType == CARD_NONE){
        Serial.println("No card attached");
        return;
    }
    if(cardType == CARD_MMC){
        Serial.println("MMC");
    } else if(cardType == CARD_SD){
        Serial.println("SDSC");
    } else if(cardType == CARD_SDHC){
        Serial.println("SDHC");
    } else {
        Serial.println("UNKNOWN");
    }
 
    uint64_t cardSize = SD_MMC.cardSize();
    int cardSizeInMB = cardSize/(1024 * 1024);
     
    Serial.print("Card size: ");
    Serial.println(cardSizeInMB);
 
 
    uint64_t bytesAvailable = SD_MMC.totalBytes(); 
    int bytesAvailableInMB = bytesAvailable/(1024 * 1024);
 
    Serial.print("MB available: ");
    Serial.println(bytesAvailableInMB);
 
 
    uint64_t bytesUsed = SD_MMC.usedBytes();
    int bytesUsedInMB = bytesUsed/(1024 * 1024);
 
    Serial.print("MB used: ");
    Serial.println(bytesUsedInMB);
}
 
void loop(){}

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13924
ho 0 tail 12 room 4
load:0x40080400,len:3600
entry 0x400805f0

Card Type: SDHC
Card size: 7580
MB available: 7574
MB used: 72

closed 2023-07-03 by s60sc