s60sc/ESP32-CAM_MJPEG2SD #238: ERROR @ motionDetect.cpp:90] motionDetect: jpg2rgb() failed

ahmed-alzayani, 2023-07-05:
my esp32 was working but now i get this massage and no display 

[1:50:51 PM Web] Connect to: ws://192.168.4.1:80/ws
[1:50:51 PM Web] Connected
[10:50:50.853 wsHandler] Websocket connection: 56
[10:50:54.876 DEBUG @ mjpeg2sd.cpp:397] ============================
[10:51:00.553 DEBUG @ mjpeg2sd.cpp:397] ============================
[10:51:01.032 DEBUG @ mjpeg2sd.cpp:397] ============================
[10:51:02.072 DEBUG @ mjpeg2sd.cpp:397] ============================
[10:51:05.510 DEBUG @ mjpeg2sd.cpp:397] ============================
[10:51:17.263 DEBUG @ mjpeg2sd.cpp:397] ============================
[10:51:21.101 DEBUG @ mjpeg2sd.cpp:397] ============================
[10:51:38.611 DEBUG @ mjpeg2sd.cpp:397] ============================
[10:51:53.883 DEBUG @ mjpeg2sd.cpp:397] ============================
[10:51:56.041 DEBUG @ mjpeg2sd.cpp:397] ============================
[10:51:58.901 updateAppStatus] Enabling motion detection
[10:52:01.216 savePrefs] Saved preferences
[10:52:01.234 saveConfigVect] Config file saved
[10:52:04.868 savePrefs] Saved preferences
[10:52:04.885 saveConfigVect] Config file saved
[10:52:06.133 doRestart] Controlled restart: user requested restart
[10:52:06.916 ERROR @ motionDetect.cpp:90] motionDetect: jpg2rgb() failed
[10:52:06.928 DEBUG @ mjpeg2sd.cpp:397] ============================
[1:52:51 PM Web] Disconnected: 1006 -

s60sc, 2023-07-05:
caused by bad jpeg from cam, only a problem if it occurs constantly

closed 2023-07-09 by s60sc

ahmed-alzayani, 2023-07-10:
so i have to change the cam???? or what???

s60sc, 2023-07-10:
if it was working and now its is not then it is a hardware problem.

Zickezack, 2024-01-11:
I suggest that it is the sd card which is not good anymore (limited number of writes) or due to rising ambient temperatures if you did not change anything in the settings. In my case, I got it when I tried to up the resolution to UXGA. I now went down to the standard setting of SVGA and it seems fine again. Maybe due to not enough heap...?

castruck, 2024-07-24:
I got the same error when I changed the resolution to UXGA.  It went away when i set the resolution to SXGA without rebooting.