s60sc/ESP32-CAM_MJPEG2SD #243: Trying to interface NRF24L01 with ESP32 camera

maximagination1, 2023-07-13:
Hi. I have an XIAO ESP32S3 Sense camera dev board which I want to use your code with to display First-Person-View footage from and have an NRF24L01 Radio receiver module taking input from an Arduino transmitter to drive a couple of motors. Is this possible?

Considering I have to make a few changes in the code to get the NRF24L01 working, is it technically possible to wire this SPI-based transceiver to a small board like the XIAO ESP32S3 Sense? If so, what pins would I be able to use? 

Thanks.

s60sc, 2023-07-13:
I dont have either of these modules so can only theorise.
NRF24L01 requires 5 IO pins, the XIAO appears to have 6 free pins (1-6),   SD uses 7-9. Serial uses 43, 44.
Configure SPI to use 3 of the free pins as MOSI, MISO, SCK. Other 2 pins (CSN, CE) just IO.
[image]

maximagination1, 2023-07-13:
Ok thanks, but how and where would I configure 3 other pins for SPI considering that 8, 9, and 10 are designed for that purpose?

maximagination1, 2023-07-13:
Quoted from: https://forum.arduino.cc/t/change-or-define-new-spi-pins/514335/2
"When a device's Slave Select pin is low, it communicates with the master. When it's high, it ignores the master. This allows you to have multiple SPI devices sharing the same MISO, MOSI, and CLK lines"
Based on that, in theory, could I wire the NRF24L01 module's 3 SPI pins to the XIAO's 3 labeled pins on the right and just use two of the other 6 available ones for CSN and CE?

s60sc, 2023-07-14:
No, pins 7-9 are used for the sd_mmc interface, not spi. You define the spi pins to use when you call whatever library you are using for the NRF24L01

maximagination1, 2023-07-14:
I've made changes according to your above suggestions and changed some values in the library I'm using (in txt file) - RF24. I uncommented "#define SOFTSPI" and changed SPI pins to:
#define SOFT_SPI_MISO_PIN 3
#define SOFT_SPI_MOSI_PIN 4
#define SOFT_SPI_SCK_PIN 5

Also added the necessary "#include <DigitalIO.h>" library in the code according to: https://nrf24.github.io/RF24/md_docs_arduino.html
I've tried using separate receiver test codes with the board but still does not work. I don't know what to do...

My NRF24L01 module works flawlessly with all of my other Arduino boards but not the XIAO ESP32S3...

Any possible suggestions? I know this problem is not entirely related to your code, but since I don't know who else to turn to about this issue, I thought you might know. Thanks for your help so far.

s60sc, 2023-07-14:
You need to refer the [esp32 example](https://nrf24.github.io/RF24/md_docs_arduino.html#:~:text=%7D-,ESP32%20example,-See%20also)
The modified example should look like:
```
#include <SPI.h>
#include <RF24.h>
 
// custom-defined pins
#define MY_MISO 1
#define MY_MOSI 2
#define MY_SCLK 4
#define MY_SS   5  // pass MY_SS as the csn_pin parameter to the RF24 constructor
 
RF24 radio(6, MY_SS); // the (ce_pin, csn_pin) connected to the radio
 
SPIClass* hspi = nullptr; // we'll instantiate this in the `setup()` function
 
void setup() {
  Serial.begin(115200);
 
  hspi = new SPIClass(HSPI); 
  hspi->begin(MY_SCLK, MY_MISO, MY_MOSI, MY_SS);
 
  if (!radio.begin(hspi)) {
    Serial.println("radio hardware not responding!!");
    while (1) {} // hold program in infinite loop to prevent subsequent errors
  }
 
  // ... continue with program as normal (see library examples/ folder)
}
```

Pin 3 not used as the Sense schematic shows it connected to the sd card.
Note that the NRF24L01 library may not be compatible with ESP32S3

closed 2023-07-28 by s60sc