s60sc/ESP32-CAM_MJPEG2SD #245: Possibility for x2 http Web Servers running off the same ESP32?

maximagination1, 2023-07-25:
Is it possible to have the ESP32-CAM_MJPEG2SD code run alongside another code for operating an RC Car via WiFi on its own web server and the camera streaming to a separate web server happening all on one ESP32? 

I am making an FPV RC Car with a XIAO ESP32S3 camera board as the microcontroller mainly running your code. I'd like to have the camera stream to the camera web server you developed as well as have control of the car from another phone via a separate web server (which I already got working separately).

If this is possible, would you know how I can pull this off? Would I need to add code for separating these two events on different ports and IPs? Maybe I need to also add a specific library to do this?

Thanks in advance.

s60sc, 2023-07-25:
Much simpler and more performant to host the other app on a separate ESP32, such as a cheap ESP32-C3.

maximagination1, 2023-07-25:
Thanks for your reply. I just got my other (car control) application programmed and working on an ESP32C3 today, as you suggested. Will run your camera web server separately on the ESP32S3 to another device for video feed. This really helped guide me the right way. Thanks.

closed 2023-08-03 by s60sc