s60sc/ESP32-CAM_MJPEG2SD #246: install ssh-server to enable reverse ssh tunneling

CazYokoyama, 2023-07-26:
meye camera may be located behind NAT. To connect to the meye camera, we need reverse ssh tunneling. I am not 100% sure that it can be achieved when ssh-server is installed. It may be achieved by
- run "ssh -fN -R 7010:localhost:80 YourUsername@IpAddressOfYourWifiRouter"
- browse "http://localhost:7010".
I am glad if you propose a better solution.

s60sc, 2023-07-26:
not viable for a little esp32 to be both a camera and ssh server

CazYokoyama, 2023-07-26:
> not viable for a little esp32 to be both a camera and ssh server

While you maybe correct. I'll get back after running
https://github.com/ewpa/LibSSH-ESP32/tree/master/examples/samplesshd-kbdint

closed 2023-07-27 by CazYokoyama