s60sc/ESP32-CAM_MJPEG2SD #247: A serial exception error occurred: Cannot configure port, something went wrong.

NobiteK, 2023-07-27:
Hi. I'm trying to flash ESP32-CAM AI-Thinker with MB Micro USB adapter, and I'm getting:

A serial exception error occurred: Cannot configure port, something went wrong. Original message: PermissionError(13, 'A device attached to the system is not functioning.', None, 31)
Note: This error originates from pySerial. It is likely not a problem with esptool, but with the hardware connection or drivers.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html
Failed uploading: uploading error: exit status 1

I tried holding IO0 and RST pin while plugging it in or just clicking them, and I tried reinstalling or updating CH341SER driver but it didn't worked. Please help.

s60sc, 2023-07-27:
Are you using the ESP32-CAM that came with the MB Micro USB adapter, because ESP32-CAM bought separately look the same but are not pin compatible

NobiteK, 2023-07-27:
> 

Yes, I'm using MB adapter which was along with my ESP32-CAM

s60sc, 2023-07-27:
Is it plugged in the right way round, and is the pin next to the led marked GND/R

NobiteK, 2023-07-27:
> Is it plugged in the right way round, and is the pin next to the led marked GND/R

Yes, it is plugged the right way I think.
[image]

CazYokoyama, 2023-07-27:
Are you sure that the supplying voltage is OK? When I connect the board by an 2 meter long USB extension cable, esp32cam is powered by 4.6V or so. The connection is unstable and lsusb sometime reports there is the board and sometimes not.

NobiteK, 2023-07-28:
> someone else had the same [problem](https://stackoverflow.com/questions/76146837/a-fatal-esptool-py-error-occurred-cannot-configure-port-permissionerror13-a#:~:text=I%20was%20trying,make%20it%20work.)

I managed to flash it on my Windows 10 laptop but after flashing and rebooting in serial monitor I keep getting this error:

rst:0x10 (RTCWDT_RTC_RESET),boot:0x3b (SPI_FAST_FLASH_BOOT)
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
ets Jul 29 2019 12:31:46

s60sc, 2023-07-28:
Try flashing example blink sketch

NobiteK, 2023-07-29:
> Try flashing example blink sketch

It's working. I tried flashing new clean GitHub code just with my Wi-Fi credentials and in serial monitor at first after flash it gives me IP and everything but 192.168.4.1 do not load, so I rebooted it, and it gives me the same error (invalid header: 0xffffffff).

closed 2023-08-03 by s60sc

NobiteK, 2023-08-03:
So no help?