s60sc/ESP32-CAM_MJPEG2SD #249: Initial boot observations (downloading config.txt, deleted MJPEG2SD.html)

FoxWhiskey, 2023-07-30:
Hi,

I have followed instructions of the README.md and a YoutTube-video, but probably I misunderstand the concept of the initial boot sequence.

What I have done:
1. modified `utils.cpp` to set my router's SSID, password and also set `hostName[]` (line 25)
2. copied `common.js`and `MJPEG2SD.html` onto an µSD-card
3. successfullly compiled and downloaded the firmware onto the AI-thinker ESP32 cam module
4. disconnected power/USB and inserted µSD-card in cam module
5. powered on

What I notice:   

on initial boot, the system downloads `config.txt` from github and saves it (unmodified) onto µSD-card. Then it deletes (only) the HTML-file, performs a controlled restart due to _config file download_ and consequently looses my Wifi credentials (as `config.txt` does not contain any specific Wifi-information).   
The system then boots into AP-mode, but the HMI is inaccessible as the HTML-file was deleted...

Also the set hostname in `utils.cpp` is not considered - neither before, nor after the controlled restart. It seems to be set to a default value, based on the MAC-address.

What I expect:
* the system does not loose WiFi-credentials that have been compiled into the code
* Wifi-credentials are not being read from SD-card, if compiled in code
* the `hostName[]`-string is used at all times if variable is not empty

Can you please shine some light on me, what is supposed to happen or what I'm misunderstanding?
Thanks for your effort

On a side note - I am using PlatformIO as IDE - don't know whether or not this is an important piece of information...

s60sc, 2023-07-30:
Don't change the code, let the app do the setup then use the web interface to change the settings

FoxWhiskey, 2023-07-30:
Yup. You are right.   
I think the video source I used to learn about this project is a bit _misleading_ regarding this matter: don't change code! Just use the HMI.

closed 2023-07-30 by FoxWhiskey