s60sc/ESP32-CAM_MJPEG2SD #250: Framesize always SVGA regardless of config settings

OldYin, 2023-07-30:
Changes in file: ESP32-CAM_MJPEG2SD.ino may have caused a minor issue.

The order of ...

  // Load saved user configuration
  loadConfig();

  // initialise camera
  prepCam();

has recently changed. 
As a result, the line...       

      //drop down frame size for higher initial frame rate
      s->set_framesize(s, FRAMESIZE_SVGA);

in prepCam(); now overrides the framesize defined in any configuration after a reboot.

swapping the order of  loadConfig(); &  prepCam(); back again corrects this.

s60sc, 2023-07-30:
Thanks, I'll investigate

s60sc, 2023-07-30:
Should be fixed

OldYin, 2023-07-30:
As far as the issue I was having with the framesize, that does indeed appear to be sorted. Thank you :)

closed 2023-07-30 by OldYin