s60sc/ESP32-CAM_MJPEG2SD #251: Anyone get SD card working for Seeed Xiao ESP32S3 Sense?

jlutzwpi, 2023-07-30:
I get the following after uploading the sketch on the Seeed Xiao ESP32S3 Sense:

```
[00:28:26.900 setup] =============== Starting ===============
[00:28:26.928 ERROR @ utilsFS.cpp:69] SD card mount failed 
[00:28:26.929 loadConfig] Load config
[00:28:26.929 ERROR @ prefs.cpp:133] Failed to load file /data/configs.txt 
[00:28:26.929 WARN[00:28:30.133 startPing] Started ping monitoring
[00:28:30.134 startWebServer] Starting web server on port: 80
[00:28:30.135 ERROR @ ESP32-CAM_MJPEG2SD.ino:136] Startup Failure: Check SD card inserted 
[00:28:35.133 resetWatchDog] WatchDog started using task: ping
```

The SD card works fine with other programs that I have uploaded, and I have saved data and jpegs to the 16GB SD card without issues.  However, I noticed this sketch uses SD_MMC while the sketches that I have used just use SD.

Has anyone had any success with the Xiao ESP32S3 Sense?  Just to be sure, I reformatted the card as a FAT32 using SD Card Formatter (doing a full format, and not just a quick one).

s60sc, 2023-07-31:
See #241

jlutzwpi, 2023-07-31:
Thanks for the quick feedback.  This is what the appsGlobal.h file looks like for me:

```
//#if defined(CONFIG_IDF_TARGET_ESP32)
// default pin configuration for ESP32 cam boards
//#define CAMERA_MODEL_AI_THINKER // Has PSRAM  
//#elif defined(CONFIG_IDF_TARGET_ESP32S3)
// default pin configuration below for Freenove ESP32S3 cam boards
//#define CAMERA_MODEL_ESP32S3_EYE // Has PSRAM
//#endif

/**************************************************************************
 Uncomment one only of the camera models below if not using a default above
 and comment out above define block
 Selecting wrong model may crash your device due to pin conflict
***************************************************************************/

//#define CAMERA_MODEL_WROVER_KIT // Has PSRAM
//#define CAMERA_MODEL_ESP_EYE // Has PSRAM
//#define CAMERA_MODEL_M5STACK_PSRAM // Has PSRAM
//#define CAMERA_MODEL_M5STACK_V2_PSRAM // M5Camera version B Has PSRAM
//#define CAMERA_MODEL_M5STACK_WIDE // Has PSRAM
//#define CAMERA_MODEL_M5STACK_ESP32CAM // No PSRAM
//#define CAMERA_MODEL_M5STACK_UNITCAM // No PSRAM
//#define CAMERA_MODEL_TTGO_T_JOURNAL // No PSRAM
#define CAMERA_MODEL_XIAO_ESP32S3 // Has PSRAM
//#define CAMERA_MODEL_ESP32_CAM_BOARD
//#define CAMERA_MODEL_ESP32S2_CAM_BOARD
//#define CAMERA_MODEL_ESP32S3_CAM_LCD
//#define CAMERA_MODEL_TTGO_T_CAMERA_PLUS
```
I still get the SD card mount failed error.  Also, I tried just running the simple SDMMC_Test sketch and I get a similar error.  I'm wondering if there is an issue with the board because the SD library runs fine.  I even soldered the two pads as described [here](https://wiki.seeedstudio.com/xiao_esp32s3_sense_filesystem/).  I asked a question on the Seeed Studio forum but I'm not sure how responsive they are.  Thanks again for your quick reply.

jlutzwpi, 2023-07-31:
Argh, I needed to update my arduino-esp32 in the boards manager.....I figured since I didn't get any compile errors that I was in the clear....nope!  Thanks for the quick replies.  Looks good so far!  Great work!

closed 2023-08-03 by s60sc