s60sc/ESP32-CAM_MJPEG2SD #253: Only able to run with Serial monitor opened?

jlutzwpi, 2023-08-01:
Hello again, really enjoying this code.  It works great.  Seeing 20 fps being recorded to SD.  However, it looks like the webserver only starts up when a Serial monitor is opened.  I would like to try to a battery-powered solution where I'm not connected via USB-C with a Serial monitor open.  Is there a way to disable that?  Typically it's a while(!Serial) call but there is a whole logging system so not sure if a similar command is issued elsewhere.  Thanks!

s60sc, 2023-08-01:
Nothing to do with app. Hardware issue?

s60sc, 2023-08-01:
Appears to be an [S3 specific problem](https://www.esp32.com/viewtopic.php?t=31806)
try
```
#if !CONFIG_IDF_TARGET_ESP32S3
      Serial.flush();
#endif
```

jlutzwpi, 2023-08-01:
Huh, how about that?  I loaded the MCU into my model rocket payload already, so I'll give that a try post-launch.  Thanks for digging that up. I just commented it out for the time being since I didn't really need that debug capability if I'm running battery-powered.  Thanks again!

s60sc, 2023-08-03:
new 8.7.4

closed 2023-08-03 by s60sc