s60sc/ESP32-CAM_MJPEG2SD #254: Continous loss of WiFi-connection

FoxWhiskey, 2023-08-01:
I propose to reopen issue #221 

I'm observing exact the same problem. If further information or tests are required, I'm willing to help...

s60sc, 2023-08-01:
OK. If someone determines a solution, I'll incorporate it.

Lee1312, 2023-08-13:
Yeah watched MaxImagination video, set everything up and its running fine until Wi-Fi ping starts and it breaks. Sometimes video doesn't even show. First I though it was because of Save Capture enabled, so i disabled it but the same thing happens! Here are the logs:

[00:00:22.429 startWifi] Wifi stats for Baka_Coka - signal strength: -48 dBm; Encryption: WPA2_PSK; channel: 3
[00:00:22.430 resetWatchDog] WatchDog started using task: ping
[00:00:24.128 fileHandler] Sent /data/MJPEG2SD.htm to browser
[21:52:13.000 showLocalTime] Got current time from browser: 13/08/2023 21:52:13 with tz: GMT0
[21:52:23.662 fileHandler] Sent /data/MJPEG2SD.htm to browser
[21:52:29.687 WARN pingTimeout] Failed to ping gateway, restart wifi ... 
[21:52:29.687 setWifiSTA] Wifi Station IP from DHCP
[21:52:29.691 onWiFiEvent] WiFi Station disconnected
.[21:52:29.847 onWiFiEvent] WiFi Station connection to Baka_Coka, using hostname: ESP-CAM_MJPEG_BC994F9EF0C8
.........[21:52:34.705 WARN startWifi] SSID Baka_Coka WL_IDLE_STATUS 
[21:52:38.528 startWifi] Wifi stats for Baka_Coka - signal strength: -51 dBm; Encryption: WPA2_PSK; channel: 3

Will be using the default WebCamera example, as with packet loss it does not brake and continues to transmit video, while the current code just breaks and freezes/stops transmitting. And it seems based on the logs to constantly connect and disconnect, i suppose this is the error. 

PS. One final thing i want to mention, when you transfer the 'data' folder to the SD card prior to Wi-Fi connection and then you connect to ESP32-Cam while it is in AP mode to once again set up the Wi-Fi credentials (which doesn't make any sense to me to do it twice, since it was already done in 'utils.cpp'). After writing the correct Wi-Fi credentials and connecting now when ESP32-Cam is in Slave/Station mode. But while updating the config files in the 'data' folder on the ESP32 SD card, it deletes the 'MJPEG2SD.htm'. So you have to go back and manually insert the file once again in the SD-Card \data\ folder for it to work kinda properly. 

I really hoped this would work much better and i will be looking out for this project as it has some nifty features! Will be using the standard example code as such as i need LiveStream video feed with minimal delay for my bachelor! Hope my input helps in solving the problem/s!

Lee1312, 2023-08-13:
One thing just to add that this does seem to happen at random intervals and isn't happening every time. In my case video connects sometimes, other times it doesn't, sometimes it crashes sometimes it doesn't. Really random stuff...

s60sc, 2023-08-21:
[FoxWhiskey](https://github.com/FoxWhiskey) try 8.7.5 with `Use_ping` set to off

FoxWhiskey, 2023-08-25:
> [FoxWhiskey](https://github.com/FoxWhiskey) try 8.7.5 with `Use_ping` set to off

Will do so. But your request comes at an _unfortunate_ point of time, as I'm currently spending my time with the family summer holidays. I have smuggled all equipment necessary to compile and execute code into my luggage, but I will have to wait for the right moment ;-)

aklstreak, 2023-09-05:
My observation:
it seems that the ESP32-Cam becomes unresponsive to the call for the main website only. The module is still connected to WiFi and also responds to commands though.

I am using an ESP32-Cam in my home automation platform using mqtt.  I am pulling a jpg every couple of seconds into webpage via a http request. That keeps working even if I can't reach the module' webpage. The same with other commands sent to the module via mqtt. Switching on the LED, switching on/off motion detection etc all keep working fine.

So far I have to power cycle the module to gain access to the web page again.

Zeunas, 2023-09-18:
> My observation: it seems that the ESP32-Cam becomes unresponsive to the call for the main website only. The module is still connected to WiFi and also responds to commands though.
>
> I am using an [...]

Sorry for high jacking this topic but how did you set the MQTT up on both the ESP-cam and HA? I have mosquitto set up with user and password, I added the same details on esp-cam and although mosquitto recognises the connection it doesn't authorise it...I'm pulling my hair already as I can't find a solution.

As for the wifi connection issue, I did have this right at the beginning when I set the esp-cam up and I think it might be a memory + power issue. The power issue is easy to fix but the memory issue is a bit of a pain...on my set up I don't actually need it and turning it off fixed the issue (I sometimes turn it back on and connection stays stable but haven't left it on for long enough to lose connection). Sorry for not being much of help.

closed 2023-10-12 by s60sc